Twilight-Logic/AR488 #7: two questions 1.atmega168 ok? 2 only use atmega328p/168p direct to rs232 with max232/max3232?

youxiaojie, 2019-09-25:
two questions 1.only use atmega328p/168p direct to rs232 with max232/max3232?
detailed :
1.does atmega 168 works ok? 
2.I burn atmega 328/168 with ardunio uno and move to a pcb with power supply, max232/max3232 rs232 TTL voltage converter and then connect COM1 port of computer, and ardunio connect to gbip?
3. I see almost full of converter has sn7516x as isolation why there is no such things?Is it easy to damage atmega328p?
4. If I use another arduino but some pin is occupied by led, can I change the pin?
thanks!

Twilight-Logic, 2019-09-29:
Thank you for your questions. I hope the information below answers at least some of your points.

0.does this support labview?
A. I have not tested it with Labview as I do not have and cannot afford the software, but Labview  does support Prologix interfaces. I noted from a couple of forum posts that you need to send CRLF and have ++auto 1 turned on. I would be interested to know myself whether Labview works with the AR488 using the Prologix driver.

1.only use atmega328p/168p direct to rs232 with max232/max3232?
1.does atmega 168 works ok?
A. The firmware has been tested with the Uno (328p) and Mega 2560. I see no reason why it should not work with a 328p IC with an external MAX232/MAX3232 chip, provided that the signal voltage levels between the two chips are compatible. The cheap clones use a CH340 UART and I would expect a MAX UART to work in a similar way. At present I do not have the means to test the 168p, although it does seem to be pin compatible with the 328p. If you happen to try it, then I would be interested in your results. I will see whether I can procure a development board to test it. If you do happen to test it, then please could you let me know your findings?

2.I burn atmega 328/168 with ardunio uno and move to a pcb with power supply, max232/max3232 rs232 TTL voltage converter and then connect COM1 port of computer, and ardunio connect to gbip?
A. I see no reason why it shouldn't communicate via the COM1 port of a computer, although the maximum achievable baud rate may be lower than that which can be emulated over USB. If by 'arduinio' you mean the 328p (or 168p) then yes, the appropriate 328p GPIO pins would need to be somehow connected to the GPIB bus. In principle, it looks like your idea should work and I will be interested to know how your project turns out.

3. I see almost full of converter has sn7516x as isolation why there is no such things?Is it easy to damage atmega328p?
A. I would not advise running more than 2 or 3 devices from it due to the current handling of the GPIO outputs, however, in a 1:1 scenario there should be no problem. So far, I have had no reports of damage to the GPIO outputs, but of course, everyone has a different level of skill and confidence, so it is impossible to rule out ANY form of damage, but provided that everything is wired correctly, there should be no problem. However, please read the section 'Multiple Arduinos on the bus and problems with instruments in the manual. There are two reasons for the lack of SN7516x type isolation: firstly, a shortage of pins on the 328p, and secondly additional cost. Having said that, someone did mention about developing a shield for that. Now that the code has been ported to the Mega (with other boards possibly to follow) it opens up further possibilities which could be developed in time.

4. If I use another arduino but some pin is occupied by led, can I change the pin?
thanks!
A. You could remove the LED or, in version 0.47.xx of the firmware there will be an option to specify a custom layout so you could use another pin.

5 atmega 32u4 has internal usb can it be modified to use this?(use lufa get a serial port?)
A. Interesting thought. It has already been mentioned by artag, who has successfully experimented with using that board (for the code see contributors on the Github). It seems that it worked despite the 3.3v output with 1 device on the GPIB bus, but he hasn't implemented that particular feature yet. I have just received a couple of 32u4 boards to experiment with myself and it is something I would like to investigate in due course.

Twilight-Logic, 2019-10-13:
Atrag's 32u4 port has now been merged into the codebase and the Micro is now supported. The manual has been updated with the pinout for that board. It currently employs serial over USB.

youxiaojie, 2019-10-15:
that is great!!

Twilight-Logic, 2019-10-25:
No further comments so issue closed.

closed 2019-10-25 by Twilight-Logic