Twilight-Logic/AR488 #9: Arduino Pro Micro (32U4 chinese clone) works only if USE_SERIALEVENT is not defined

krasimirk, 2020-01-22:
Hello,
I tried to run the AR488.ino on Arduino Pro Micro, but I do not get any answer from the FW.
After the debug I found that if I comment out USE_SERIALEVENT, the code is working.
This is not request for bug fixing, just a tip for workaround if somebody have the same problem.
I do not have time to investigate why the original code does not work.
Regards,
Krasimir Kostadinov

Twilight-Logic, 2020-01-22:
Krasimir, thank you for your feedback. After a little online investigation is would seem that unfortunately SerialEvent is not available on the Leonardo, Micro or other 32u4 boards:

https://www.arduino.cc/en/Tutorial/SerialEvent

I presume that this is because the interrupt is not generated by CDC serial ports. The above link is the only mention I could find in the official documentation, and even then, the mention is found only within the comments of the provided code example! It was mentioned in a couple of forums though.

I will amend the code in the next release to take into account the lack of support for this interrupt on 32u4 boards.

krasimirk, 2020-01-23:
Welcome!
I tested successfully HP 34401 with basic SCPI commands for measurements, mode settings etc.
Thank you for spending your time to make available this very useful FW!
Let me know if you need support for testing it for different DMMs. I own HP 34401/Advantest R6581T at home and I have access to Keithley 2000, 2002, Rigol DM3068 at work. A friend of mine has HP 3478 as well.
I made the "artag" version of the AR488 PCB. 
Regards!

Twilight-Logic, 2020-01-24:
I am glad that worked Ok. Thank you also for your offer of support for testing on various instruments. Are any of your instruments capable of using secondary addressing? I already know that the Keithley 2000 iand the HP3478 are not. Just wondering about the others you mention?

Tha "artag" version was actually contributed by member Artag on the EEVblog forum. He was particularly keen to use the Pro Micro board because it is nice and compact.

krasimirk, 2020-01-24:
Hello,
According K2002 : " Note, however, that many devices, including the Model 2002, do not use secondary addressing."
In Hp 34401 manual, I can see only that addresses can be set between 0 and 31 (31 is talk only addrress) which are primary addresses.
For R6581T I can check on the real DMM today later. Documentation is in Japanese only and my wife and daughter reject to help me to translate it to English :-(.
Have a nice day.
P.S. : you can close the issue and contact me at krasimir.k at gmail.

Twilight-Logic, 2020-01-24:
The reason I asked is because the original Prologix did provide a secondary addressing feature. Since I didn't have an instrument that supported it, I couldn't implement and test it. So far though, no one whom I have asked so far has an instrument that actually made use of secondary addressing either. I'm not sure what, if anything actually uses it!

Since the problem has now been addressed I will now close this issue.

Full details of workaround for firmware 0.48.03 here:
https://www.eevblog.com/forum/projects/ar488-arduino-based-gpib-adapter/msg2882824/#msg2882824

The problem is fixed in firmware version 0.48.07.

closed 2020-01-24 by Twilight-Logic