Twilight-Logic/AR488 #10: Compile Fails when USE_MACROS is defined (ver 0.48.24)

1xrtt, 2020-05-09:
Hello WaveyDipole, greetings from a felow eevblogger!
I downloaded version 0.48.24 and tried to compile it with USE_MACROS defined, but it fails, giving an error about processLine(pBuf, pbPtr, 1) not defined.
Indeed, the function is not present in the code, so I searched through the archives and find it on  0.46.32 code, so I simply copy/pasted it to 0.48.24. After that, it compiled, uploaded ok and I could test it ok with MACRO_0 defined.
I´m lazy enough to no check if there is any update on this function, between these versions, but maybe you want to double check on that.

Thank you for the awesome job on creating this code, I have four adapters built, two using Nanos and two using the LGT8F328P clone and now got all of them upgraded to latest code.

Best Regards,
1xrtt

Twilight-Logic, 2020-05-12:
Thanks for reporting the issue. I will check it out and report back.

Twilight-Logic, 2021-01-10:
Was fixed in version 0.48.28.

closed 2021-01-10 by Twilight-Logic