Twilight-Logic/AR488 #11: Can't break out of read with ++ command

dgesswein, 2020-11-03:
If you have a read from device in progress such as ++read eoi and the device wasn't set to send eoi you can't use a ++ command to recover.

Twilight-Logic, 2020-12-18:
Thank you for your report. Please could you confirm which Arduino board you are using and what options you have set? Perhaps attach a copy of your AR488_Config.h. Thanks.

dgesswein, 2020-12-19:
Its a UNO. File attached.

#ifndef AR488_CONFIG_H
#define AR488_CONFIG_H

/*********************************************/
/***** AR488 GLOBAL CONFIGURATION HEADER *****/
/***** vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv *****/

/***** Firmware version *****/
#define FWVER "AR488 GPIB controller, ver. 0.48.29, 11/01/2020"

/***** BOARD CONFIGURATION *****/
/*
 * Platform will be selected automatically based on
 * Arduino definition.
 * Only ONE board/layout should be selected per platform
 * Only ONE Serial port can be used to receive output
 */

/*** Custom layout ***/
/*
 * Uncomment to use custom board layout
 */
//#define AR488_CUSTOM

/*
 * Configure the appropriate board/layout section
 * below as required
 */
#ifdef AR488_CUSTOM
  /* Board layout */
  /*
   * Define board layout in the AR488 CUSTOM LAYOUT
   * section below
   */
  /* Serial ports */
  #define AR_HW_SERIAL
  #define AR_SERIAL_PORT Serial
  //#define AR_SERIAL_PORT Serial1
  //#define AR_SERIAL_PORT Serial2
  //#define AR_SERIAL_PORT Serial3
  //#define AR_CDC_SERIAL
  //#define AR_SW_SERIAL

/*** UNO and NANO boards ***/
#elif __AVR_ATmega328P__
  /* Board/layout selection */
  #define AR488_UNO
  //#define AR488_NANO
  /*** Serial ports ***/
  //Select HardwareSerial or SoftwareSerial (default = HardwareSerial) ***/
  // The UNO/NANO default hardware port is 'Serial'
  // (Comment out #define AR_HW_SERIAL if using SoftwareSerial)
  #define AR_HW_SERIAL
  #ifdef AR_HW_SERIAL
    #define AR_SERIAL_PORT Serial
    #define USE_SERIALEVENT
  #else
    // Select software serial port
    #define AR_SW_SERIAL
  #endif

/*** MEGA 32U4 based boards (Micro, Leonardo) ***/
#elif __AVR_ATmega32U4__
  /*** Board/layout selection ***/
  #define AR488_MEGA32U4_MICRO  // Artag's design for Micro board
  //#define AR488_MEGA32U4_LR3  // Leonardo R3 (same pin layout as Uno)
  /*** Serial ports ***/
  // By default the CDC serial port is used
[... 262 lines cut ...]
  #ifdef DB_HW_SERIAL
    extern HardwareSerial *dbSerial;
  #endif
  // Note: SoftwareSerial support conflicts with PCINT support
  #ifdef DB_SW_SERIAL
    #include <SoftwareSerial.h>
    extern SoftwareSerial *dbSerial;
  #endif
#endif

/******^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^******/
/***** SERIAL PORT EXTERNAL DECLARATIONS *****/
/*********************************************/

/*********************************************/
/***** MISCELLANEOUS DECLARATIONS *****/
/******vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv******/

#define AR_CFG_SIZE 84

/******^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^******/
/***** MISCELLANEOUS DECLARATIONS *****/
/*********************************************/

#endif // AR488_CONFIG_H

Twilight-Logic, 2020-12-21:
Thank you. I have confirmed the issue and will update as soon as I have a solution.

Twilight-Logic, 2021-01-12:
Fixed in version 0.49.12

closed 2021-01-12 by Twilight-Logic