Twilight-Logic/AR488 #15: Normalize the git repository

douardda, 2021-05-02:
Hi,
this project looks very interesting, and I'd like to help for an ESP32 version of the code.
However a few things look a bit strange with the way the git repo is managed:

- why not use tags for published version?
- why add the zip file of the "last" version (can be omitted is properly tagged), as well as 
- why keep old version as a bunch of binary (zip) files instead of using what git is made for (using tags)?
- it would be also nice the add the sources of the documentation instead of the pdf files directly (using a [markdown](https://www.markdownguide.org/), [ReST](https://docutils.sourceforge.io/rst.html) or even a tool like [sphinx](https://www.sphinx-doc.org/) to manage the documentation).
- ideally, binary files (docs and firmwares) should be automatically built (eg. using [travis-ci)](https://travis-ci.org/)
- it would be nice to have the work in progress for other platforms (eg. esp32) available as branches.

Would you consider these improvements? Would you like some help for this?

Thank you
David

Twilight-Logic, 2021-06-03:
David, thank you for your kind offer of help. I would certainly consider such improvements and your offer of assistance is greatly appreciated. You have already mentioned a couple of tools which I will explore so thank you for this.

douardda, 2021-10-16:
Hi,

I've started doing some cleaning stuff in https://github.com/douardda/AR488/tree/normalize 
I've removed all the zip files (as well as the Contributed directory which IMHO should be merged with the main code; which I ma try to do later on).

For the doc, it would be handy to get access to the source fie of the PDF you publish, so that I can try to convert it as a sphinx project.

Also note that in my fork, I have created tags for versions I could identify in the git log.

douardda, 2021-10-16:
I've started a sphinx-based doc (copy pasting from the PDF). You can have an idea of the result here: https://sdfa3.org/david/ar488/ 

It's a straight sphinx config (default theme, nothing tuned or adapted).

Note that sphinx can also generated PDF files out of the box

douardda, 2021-10-17:
Got the main documentation converted to sphinx in [this branch](https://github.com/douardda/AR488/tree/normalize) (in the docs/  directory); the HTML result is visible on the [same URL](https://sdfa3.org/david/ar488/).

It needs some work to make it look a bit better, but it's a nice starting point IMHO.

Meanwhile, I have also played a bit with github actions to build the firmware automatically.

  https://github.com/douardda/AR488/actions 

For example, generated firmware files for my latest git revision are listed on:
 
  https://github.com/douardda/AR488/actions/runs/1351644598

For this I am using [platformio](https://platformio.org) to compile the firmware for different boards (this works in a branch of mine in which I am porting the code to work on a STM32, here a Nucleo F303K8 I had laying around, but that's another story I'll talk about later :-) )