Twilight-Logic/AR488 #16: Error compiling

Hoverhead, 2021-06-05:
This is "straight out of the box" no mods /***** FWVER "AR488 GPIB controller, ver. 0.49.14, 02/03/2021" *****/
I have a Sainsmart UNO and a duinotech screw sheild.  The IDE is 2:1.0.5+dfsg2-4.1 under Kubuntu.
Below is the error out put after a verify compile. 
I have no idea where I should start  to sort this.
Help please.

In file included from AR488.ino:13:0:
AR488_Eeprom.h: In function ‘void epViewData(T*)’:
AR488_Eeprom.h:32:12: error: ‘class EEPROMClass’ has no member named ‘get’
     EEPROM.get(addr, dbuf);
            ^
AR488.ino: In function ‘void setup()’:
AR488.ino:564:13: error: ‘flushPbuf’ was not declared in this scope
AR488.ino:603:13: error: ‘initAR488’ was not declared in this scope
AR488.ino:647:36: error: ‘initDevice’ was not declared in this scope
AR488.ino:650:40: error: ‘initController’ was not declared in this scope
AR488.ino: In function ‘void loop()’:
AR488.ino:707:24: error: ‘execCmd’ was not declared in this scope
AR488.ino:714:35: error: ‘sendToInstrument’ was not declared in this scope
AR488.ino:718:25: error: ‘gpibReceiveData’ was not declared in this scope
AR488.ino:723:25: error: ‘gpibReceiveData’ was not declared in this scope
AR488.ino:730:19: error: ‘spoll_h’ was not declared in this scope
AR488.ino:735:52: error: ‘gpibReceiveData’ was not declared in this scope
AR488.ino:741:51: error: ‘sendToInstrument’ was not declared in this scope
AR488.ino:743:15: error: ‘lonMode’ was not declared in this scope
AR488.ino:745:31: error: ‘attnRequired’ was not declared in this scope
AR488.ino:746:51: error: ‘sendToInstrument’ was not declared in this scope
AR488.ino:761:22: error: ‘serialIn_h’ was not declared in this scope
AR488.ino: In function ‘void initDevice()’:
AR488.ino:778:23: error: ‘setGpibControls’ was not declared in this scope
AR488.ino: In function ‘void initController()’:
AR488.ino:788:23: error: ‘setGpibControls’ was not declared in this scope
AR488.ino:792:9: error: ‘ifc_h’ was not declared in this scope
AR488.ino: In function ‘uint8_t serialIn_h()’:
AR488.ino:810:47: error: ‘parseInput’ was not declared in this scope
AR488.ino: In function ‘uint8_t parseInput(char)’:
AR488.ino:867:20: error: ‘addPbuf’ was not declared in this scope
AR488.ino:873:23: error: ‘flushPbuf’ was not declared in this scope
AR488.ino:874:36: error: ‘showPrompt’ was not declared in this scope
AR488.ino:882:38: error: ‘isCmd’ was not declared in this scope
AR488.ino:886:27: error: ‘flushPbuf’ was not declared in this scope
AR488.ino:892:64: error: ‘isIdnQuery’ was not declared in this scope
AR488.ino:894:25: error: ‘flushPbuf’ was not declared in this scope
AR488.ino:911:20: error: ‘addPbuf’ was not declared in this scope
AR488.ino:923:18: error: ‘addPbuf’ was not declared in this scope
AR488.ino:935:19: error: ‘isCmd’ was not declared in this scope
AR488.ino:939:17: error: ‘flushPbuf’ was not declared in this scope
AR488.ino: At global scope:
AR488.ino:1020:23: error: ‘addr_h’ was not declared in this scope
AR488.ino:1021:40: error: ‘aspoll_h’ was not declared in this scope
AR488.ino:1022:23: error: ‘amode_h’ was not declared in this scope
AR488.ino:1023:40: error: ‘clr_h’ was not declared in this scope
AR488.ino:1024:40: error: ‘dcl_h’ was not declared in this scope
AR488.ino:1025:40: error: ‘default_h’ was not declared in this scope
AR488.ino:1026:23: error: ‘eoi_h’ was not declared in this scope
AR488.ino:1027:23: error: ‘eor_h’ was not declared in this scope
AR488.ino:1028:23: error: ‘eos_h’ was not declared in this scope
AR488.ino:1029:23: error: ‘eot_char_h’ was not declared in this scope
AR488.ino:1030:23: error: ‘eot_en_h’ was not declared in this scope
AR488.ino:1031:40: error: ‘ifc_h’ was not declared in this scope
AR488.ino:1032:23: error: ‘id_h’ was not declared in this scope
AR488.ino:1033:23: error: ‘idn_h’ was not declared in this scope
AR488.ino:1034:23: error: ‘llo_h’ was not declared in this scope
AR488.ino:1035:23: error: ‘loc_h’ was not declared in this scope
AR488.ino:1036:23: error: ‘lon_h’ was not declared in this scope
AR488.ino:1037:23: error: ‘macro_h’ was not declared in this scope
AR488.ino:1038:23: error: ‘cmode_h’ was not declared in this scope
AR488.ino:1039:40: error: ‘ppoll_h’ was not declared in this scope
AR488.ino:1040:23: error: ‘read_h’ was not declared in this scope
AR488.ino:1041:23: error: ‘rtmo_h’ was not declared in this scope
[... 96 lines cut ...]
AR488.ino:2671:23: error: ‘gpibWriteByte’ was not declared in this scope
AR488.ino:2678:23: error: ‘gpibWriteByte’ was not declared in this scope
AR488.ino:2701:22: error: ‘uaddrDev’ was not declared in this scope
AR488.ino:2712:25: error: ‘setGpibControls’ was not declared in this scope
AR488.ino:2716:25: error: ‘setGpibControls’ was not declared in this scope
AR488.ino: In function ‘bool gpibReceiveData()’:
AR488.ino:2751:31: error: ‘addrDev’ was not declared in this scope
AR488.ino:2759:45: error: ‘Wait_on_pin_state’ was not declared in this scope
AR488.ino:2761:25: error: ‘setGpibControls’ was not declared in this scope
AR488.ino:2766:25: error: ‘setGpibControls’ was not declared in this scope
AR488.ino:2794:45: error: ‘gpibReadByte’ was not declared in this scope
AR488.ino:2827:42: error: ‘isTerminatorDetected’ was not declared in this scope
AR488.ino:2873:18: error: ‘uaddrDev’ was not declared in this scope
AR488.ino:2878:25: error: ‘setGpibControls’ was not declared in this scope
AR488.ino:2882:25: error: ‘setGpibControls’ was not declared in this scope
AR488.ino: In function ‘uint8_t gpibReadByte(uint8_t*, bool*)’:
AR488.ino:2963:45: error: ‘Wait_on_pin_state’ was not declared in this scope
AR488.ino:2983:46: error: ‘Wait_on_pin_state’ was not declared in this scope
AR488.ino: In function ‘bool gpibWriteByte(uint8_t)’:
AR488.ino:3007:34: error: ‘gpibWriteByteHandshake’ was not declared in this scope
AR488.ino: In function ‘bool gpibWriteByteHandshake(uint8_t)’:
AR488.ino:3027:46: error: ‘Wait_on_pin_state’ was not declared in this scope
AR488.ino:3032:47: error: ‘Wait_on_pin_state’ was not declared in this scope
AR488.ino:3044:46: error: ‘Wait_on_pin_state’ was not declared in this scope
AR488.ino:3050:47: error: ‘Wait_on_pin_state’ was not declared in this scope

Twilight-Logic, 2021-06-07:
That looks like a corrupt file (a { or } missing somewhere. I would deleting the unpacked files, uncompressing the ZIP again and compiling the freshly unpacked files. In the meantime will check it out as well when I get a chance. I do development on a Mega 2560 and Uno and then compile binaries for the other platforms so it should definitely work on a Uno, although have never tried on a Sainsmart one. Are you using the generic Arduino Uno board or have you downloaded additional boards?

Hoverhead, 2021-06-15:
Hi, I deleted the files on the puter, did a re-clone and the rest.  Still getting the same error result when I verify. Well the ^ is under the "g" in get this time instead of "E" in EEPROM.   There is no difference if the Arduino is plugged in or not.  This is the result of verify straight after loading AR488,ino into the IDE.  Should I be making other changes to the  AR488_Eeprom.h file first?  I don't know what I would need to change in this instance.
Could it be just a syntax issue on that line? 
In file included from AR488.ino:13:0:
AR488_Eeprom.h: In function ‘void epViewData(T*)’:
AR488_Eeprom.h:32:12: error: ‘class EEPROMClass’ has no member named ‘get’
     EEPROM.get(addr, dbuf);
                      ^

Twilight-Logic, 2021-06-19:
The compile process is independent of the hardware so having the Arduino plugged in vs unplugged should make no difference.

Which version of the Arduino IDE are you using?
Which board have you selected in the Boards Manager?

Try uncommenting this line in AR488_Config_h:

`//#define AR488_CUSTOM`

Does the same thing happen?

There is also no harm in commenting out the whole template definition in AR488_EEPROM.h as it is a diagnostic feature and not actually used:

```
/*
template<typename T> void epViewData(T* output) {
  uint16_t addr = 0;
  uint8_t dbuf[16];
  char cnt[4]= {'\0'};
  char oct[4] = {'\0'};

  // Read data
  memset(dbuf, 0x00, 16);
  for (addr=0; addr<512; addr=addr+16){
    sprintf(cnt, "%03d", addr);
    output->print(cnt);
    output->print(":");
    EEPROM.get(addr, dbuf);
    for (int i=0; i<16; i++){
      output->print(" ");
      sprintf(oct, "%02X", dbuf[i]);
      output->print(oct);
    }
    output->println();
  }
}
*/
```
I do suspect that the compile will fail elsewhere though instead.

Hoverhead, 2021-06-25:
Hi,
I did this a week ago now and sent the result by email. I see it hasn't linked to this thread.  
Needles to say those changed didn't work.  I'm now trying with my laptop to see it that makes a difference.

Twilight-Logic, 2021-06-25:
I set up a virtual machine and installed the Arduino IDE from the repository which installed version 2:1.0.5+dfsg2-4.1. A compile test confirmed the following compile errors:

```
In file included from AR488_Eeprom.cpp:3:0:
AR488_Eeprom.h: In function ‘void epViewData(T*)’:
AR488_Eeprom.h:32:12: error: ‘class EEPROMClass’ has no member named ‘get’
     EEPROM.get(addr, dbuf);
            ^
AR488_Eeprom.cpp: In function ‘void epWriteData(uint8_t*, uint16_t)’:
AR488_Eeprom.cpp:70:12: error: ‘class EEPROMClass’ has no member named ‘update’
     EEPROM.update(addr+i,cfgdata[i]);
            ^
AR488_Eeprom.cpp:74:10: error: ‘class EEPROMClass’ has no member named ‘put’
   EEPROM.put(0, crc);
          ^
AR488_Eeprom.cpp: In function ‘bool epReadData(uint8_t*, uint16_t)’:
AR488_Eeprom.cpp:92:10: error: ‘class EEPROMClass’ has no member named ‘get’
   EEPROM.get(0,crc1);
          ^
AR488_Eeprom.cpp: In function ‘bool isEepromClear()’:
AR488_Eeprom.cpp:112:10: error: ‘class EEPROMClass’ has no member named ‘get’
   EEPROM.get(0, crc);
```

Further investigation shows that the EEPROM .get and .put methods were not actually supported in that older version. Commenting out the relevant EEPROM section leads to further compile errors as shown above. I can therefore confirm that the problem exists when compiling using this early 1.0.5 version of the IDE and would therefore advise not using it to compile the sketch. It should be removed and purged to remove any old library files that might have been left behind:

sudo apt remove arduino
sudo apt purge arduino

The latest version of the IDE should then be downloaded from arduino.cc and used instead.

Hoverhead, 2021-06-25:
Unfortunately I had tried it on the lap top as I typed the last response.  Your latest reply has me working on the desktop on the Arduino.cc Download.  It has not gone well.  I had done the remove and purge  (and reboot) before I read your latest.  I had a "not booted in systemd" error amongst others.  Happy to know that it was the IDE version that was the issue, I only now have to get the latest IDE to play nice on my puter.

Hoverhead, 2021-06-25:
Success!  Had "System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down" after running "./arduino-linux-setup.sh  $USER" but ran "install.sh" anyway and have the IDE running and a successful verify.  Thanks for the directions.  Looking forward to getting my old HP 7470A pen plotter going again.

Hoverhead, 2021-06-26:
Now that I appear to have this running, I'm don't know how to get it to function as the interface to my plotter.   Is the information there and I haven't seen it yet?  Or am I on a search and rescue mission?

Twilight-Logic, 2022-02-02:
The compiling issue is now resolved so closing the issue.

Sending a file to a plotter should be possible with the Sainsmart board since it doesn't use the cheaper CH340 UART that the clones do which do not provide proper hardware handshaking. It ought to be a matter of just addressing the plotter in controller mode and sending data to it. The UART controls the data transfer from the PC, and GPIB handshaking should be sufficient for the plotter to request and pause data transfer from the interface. I will need to check how the plotter handles the data transfer.

Working with a plotter is not likely to be possible when working with a clone board.

closed 2022-02-02 by Twilight-Logic