Twilight-Logic/AR488 #21: device mode consult

michaelshow, 2021-11-01:
I test the device mode function of AR488:
Scenario 1: I connect AR488 and NI-ENET to the GPIB bus at the same time, AR488 as the device, and NI-ENET as the control terminal. NI-ENET lists AR488 in MAX, and AR488 and NI-ENET can read and write data with each other.
Scenario 2: NI-USB-HS as the control terminal and NI-ENET as the device. They can also read and write each other;
Scenario 3: NI-USB-HS is used as the control terminal, and AR488 is used as the device. NI-USB-HS can't find AR488 in MAX. Sending "ibwrt" after connecting directly with "ibdev" command will report an error: "EBUS".
In scenario 3, what do I need to do to make the normal communication between NI-USB-HS and AR488?

michaelshow, 2021-11-01:
I use the hardware made by arduino uno（Appendix A – Connection and technical information for Uno and Nano boards）。
Is it related to the lack of SN7516x GPIB transceiver integrated circuits?

Twilight-Logic, 2021-11-01:
That is curious and I am uncertain why your NI-USB-HS cannot see the Arduino. I imagine that you have tried resetting the Arduino? The difficulty with using an Arduino without buffering (e.g. SN7516x chipset) becomes apparent when you have two or more of them on the same bus. I would not expect it to be an issue with a single Arduino talking to NI devices.

What I suggest is to enable DEBUG5. That will show in a serial terminal the addressing and GPIB commands being passed to the AR488 from the NI-USB-HS.

Adding DEBUG3 and DEBUG7 will show further detail.

If you can send me the output at the terminal from the AR488 with DEBUG 5 enabled and then with all three enabled that may be helpful in deducing what the problem may be. GPIB control signals can occasionally get left in an unexpected state but a reset should usually be sufficient to clear that.

Twilight-Logic, 2021-11-03:
Thank you for your hard work on this. I really need to get to grips with something like Sphinx. The documentation looks great. I have also not noticed GitHub actions before. Had a quick look at it looks like there is lots to learn. Again must have a look at this.

I had intended to look at porting to the STM32 so your work on that is most welcome.

michaelshow, 2021-11-25:
Hello, I’m sorry that I didn’t reply in time because I was busy with other things in the middle.
I tried enable debug3, debug5, and debug7. The following scenes are recorded:
1. As a controller, AR488 can communicate with other devices normally;
2. AR488 is used as a device and NI-ENET is used as a controller: NI-ENET cannot recognize AR488;
3. On the basis of scenario 2, add another device (34401A addr 22) to the GPIB bus. NI-ENET can recognize AR488 and communicate, but it will also detect other devices by mistake. NI-ENET will recognize dozens of non-existent devices, you can refer to the attached pictures;
4. On the basis of scenario 3, connect NI-USB-HS as a controller and NI-ENET as a device. NI-USB-HS cannot recognize any devices.

The debug log is recorded by the serial communication tool, and you can refer to the attachment.
Looking forward to your answer, thank you！

[image]
[log_20211124.txt](https://github.com/Twilight-Logic/AR488/files/7600090/log_20211124.txt)
[log_20211125.txt](https://github.com/Twilight-Logic/AR488/files/7600091/log_20211125.txt)

Twilight-Logic, 2022-02-02:
Thank you for the detailed summary.
I am curious whether you still have the problem with the latest release - 0.51.03?
There have been some significant changes to the code.

Twilight-Logic, 2022-10-12:
Closing due to no further comments. Version 0.51.15 is now available to try if required.

closed 2022-10-12 by Twilight-Logic