Twilight-Logic/AR488 #23: After 3 command is sended trimmed string to GPIB

AidenShaw2020, 2021-11-26:
Hello, i have strange issue with AR488.
My configuration:

```
AR488_NANO
AR_SERIAL_BAUD 115200
++auto 1
++addr 5
```

When i send first 3 commands to HP7475A plotter, it is recognized and processed:

```
Initialised GPIB control mode
parseInput: Received SP0;
R: 2
BufferStatus: 2
sendToInstrument: Received for sending: 53 50 30 3B 
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Device addressed.
Set GPIB lines for writing data
Set write data mode.
**Send->SP0;<-End.**
Appended CR
Appended LF
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Unlisten done
Set GPIB lines to idle state
<- End of send.
sendToInstrument: Sent.
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for reading data
gpibReceiveData: Start listen ->
Before loop flags:
TRNb: 0
rEOI: 0
ATN:  0
0 
After loop flags:
ATN: 0
TMO:  1
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines to idle state
<- End listen.
```
```
parseInput: Received SP1;
R: 2
BufferStatus: 2
sendToInstrument: Received for sending: 53 50 31 3B 
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Device addressed.
Set GPIB lines for writing data
Set write data mode.
**Send->SP1;<-End.**
Appended CR
Appended LF
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Unlisten done
Set GPIB lines to idle state
<- End of send.
sendToInstrument: Sent.
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for reading data
gpibReceiveData: Start listen ->
Before loop flags:
TRNb: 0
rEOI: 0
ATN:  0
0 
After loop flags:
ATN: 0
TMO:  1
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines to idle state
<- End listen.
```
```
parseInput: Received SP2;
R: 2
BufferStatus: 2
sendToInstrument: Received for sending: 53 50 32 3B 
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Device addressed.
Set GPIB lines for writing data
Set write data mode.
**Send->SP2;<-End.**
Appended CR
Appended LF
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Unlisten done
Set GPIB lines to idle state
<- End of send.
sendToInstrument: Sent.
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for reading data
gpibReceiveData: Start listen ->
Before loop flags:
TRNb: 0
rEOI: 0
ATN:  0
0 
After loop flags:
ATN: 0
TMO:  1
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines to idle state
<- End listen.
```

But after 3 command, its sended only part of string (S, instead SP3;), and working again after plotter turn off/on.
```
parseInput: Received SP3;
R: 2
BufferStatus: 2
sendToInstrument: Received for sending: 53 50 33 3B 
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Device addressed.
Set GPIB lines for writing data
Set write data mode.
**Send->S<-End.**
Set GPIB lines to idle state
<- End of send.
sendToInstrument: Sent.
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines for reading data
gpibReceiveData: Start listen ->
Before loop flags:
TRNb: 0
rEOI: 0
ATN:  0
0 
After loop flags:
ATN: 0
TMO:  1
Set GPIB lines for sending a command
Set GPIB lines for sending a command
Set GPIB lines to idle state
<- End listen.
```

If i use ++auto 0, it is work without problems, but i need also use listen mode for OA; command (pen position).
Thanks for help.

AidenShaw2020, 2021-11-26:
I only add a comment.
Problem is appeared when I turn on talk mode on plotter, regardless on AR488 settings (++auto 1/0)

Twilight-Logic, 2021-11-27:
That is showing that a timeout occurred while sending which means it could not complete the three way handshake to send the character. That might have been caused by a change of state on the plotter or an intermittent wiring problem. I had a look at the manual to determine what the plotter does in TALK mode and I can see that OA is one of those commands sends a response to, in this case by sending its pen position. However the output above does not show the OA command being sent so it didn't get that far.

Incidentally, I notice an example in the 7475 Interfacing and Programming Manual for the Tektronix 4051 so I might just have to get one of these for my 4051 so that I can output some of the demo drawings! On the other hand there is an emulator program one can use on the PC which will take up less space....

AidenShaw2020, 2021-11-27:
OA command is not see, because i didnt send it in this example, but it is work OK (as other commands). Only thing with problem is SP commands, thats invoke communication freeze, when HP7475 is talker/listener mode.

ithinkido, 2021-12-17:
@AidenShaw2020 are you sending the `SP..`  commands all together, or are you sending them singularly with a delay between each one? Can you send large files to the plotter without the timeout issue ? I have attached a larger test file.
[Lorenz_attractor_A4.hpgl.txt](https://github.com/Twilight-Logic/AR488/files/7734944/Lorenz_attractor_A4.hpgl.txt)

AidenShaw2020, 2021-12-17:
I created python script, that fill a buffer with plt data, send it to plotter, and wait for command is finished, then send next buffer. 

Stuck for SP command is appear in both cases (with/without) delay.

ithinkido, 2021-12-17:
What happens when you send the file without the script ? Buffer overflow ?

AidenShaw2020, 2021-12-18:
yes

Twilight-Logic, 2021-12-18:
Could I get a copy of your python script please?
I don't have a plotter, but I would like to test it to see how the interface responds.

The Arduino serial port has only a small 64 byte buffer. The firmware uses a 128 byte buffer, however if you are using a clone board with a CH340 serial IC, then there is no serial handshaking so you end up with buffer overflows unless you control the data flow. The genuine Arduino Uno/Nano boards with FTDI support do not have this problem. 

The script probably needs to send the data in chunks of no more than 64 bytes with a suitable delay in between.

AidenShaw2020, 2021-12-18:
Script is ok, its same if i send only SP command manualy (with or without delay).

ithinkido, 2021-12-18:
> The Arduino serial port has only a small 64 byte buffer. The firmware uses a 128 byte buffer, however if you are using a clone board with a CH340 serial IC, then there is no serial handshaking so [...]
Sorry to hijack this issue, but the discussion here could answer a question I have had for a long time. 
@Twilight-Logic Does the ATmaga32u4 version of this project handle serial handshake ?

Twilight-Logic, 2022-02-02:
AidenShaw2020, could you let me know whether you are getting the same problem with the new version (0.51.03) please? There have been some significant changes, although this may be an issue of handshaking from the plotter side which the interface is not yet able to handle.

AidenShaw2020, 2022-02-04:
Hi Twilight-Logic, I confirmed, now its commands working properly.

closed 2022-02-04 by AidenShaw2020

Twilight-Logic, 2022-02-04:
That's good to know. Thank you.