Twilight-Logic/AR488 #25: Compile error, Mega 2560 + Arduino IDE 1.8.16

FannyLoi, 2022-02-12:
Hello,
     When attempting a first-time compile with MEGA 2560 board selected, I receive this error.  I looked through the documentation and prior issues, but was unable to find a match for what I'm facing.  Thanks very much in advance!

AR488_Layouts.cpp:1241:3: error: 'mcpPinAssertedReg' was not declared in this scope
   mcpPinAssertedReg = 0;
   ^~~~~~~~~~~~~~~~~

AR488_Layouts.cpp:1310:7: error: 'mcpPinAssertedReg' was not declared in this scope
       mcpPinAssertedReg = mcpByteRead(MCPINTCAPA);
       ^~~~~~~~~~~~~~~~~
AR488_Layouts.cpp:1314:39: error: 'gpibsig' was not declared in this scope
       return (mcpPinAssertedReg & (1<<gpibsig));
                                       ^~~~~~~
exit status 1
'mcpPinAssertedReg' was not declared in this scope

FannyLoi, 2022-02-12:
I found the directive that was indicating use of an MCP chip; it seems the checked in code has this enabled for the MEGA board.  Thanks.

closed 2022-02-12 by FannyLoi

Twilight-Logic, 2022-02-12:
Thank you for reporting this. The latest code has support for MCP23x17 chips and this was developed on a Mega board. The setting may have been accidentally left enabled. I will check and correct as necessary.

FannyLoi, 2022-02-12:
Thanks for the follow-up, I will grab the latest code!