Twilight-Logic/AR488 #26: AR488 was not declared in this scope when enabling SN7516X support

Mario-H, 2022-05-02:
Hi,

Build myself a controller with an ATMEGA328P chip and FR232R module and the 2 SN7516X transceivers.
Using IDE 1.18.15 (Win10) I can compile and upload the sketch just fine, but when I enable SN7516x support bij changing the relevant config lines to:

```
#define SN7516X

#ifdef SN7516X
 #define SN7516X_TE 6
 #define SN75161_DC 13
//  #define SN7516X_SC 12
#endif
```

I get a compile error: "AR488 was not declared in this scope" in line 511 of AR488.ino
`if (AR488.cmode==2) {`

(I did change the pin number for TE from 7 to 6 because I built the controller based on the schematic in the manual that uses pin 6 for TE and 13 for DC, but when I changed TE back to 7 I got the same compile error)

Am I missing something that I should configure?

Thanks,

Mario

Twilight-Logic, 2022-05-02:
Mario, thank you for your feedback.

Change line 511 from:

`  if (AR488.cmode==2) {
`
To:

`  if (gpibBus.cfg.cmode==2) {
`
It should then compile.

Please let me know if this fixes it.

Mario-H, 2022-05-02:
Hi,

With a 16 minutes response time that FIXED the problem you shouldn't worry about missing this.
Thanks!

Mario

Twilight-Logic, 2022-05-02:
Glad that fixed it. I will now update the code in the repository.

Mario-H, 2022-05-02:
While you're at it:
I can't get my HP8591A analyzer to respond on the commands I issue.
So I enabled all debug options hoping to get some insight in the problem.

Enabling DEBUG_IDFUNC generated another compile error in line2299 of AR488.ino :
`Found : in nested-name-specifier, expected '::'`

Is there any troubleshooting info in case the controller isn't able to control a device?
I now see a debug message:
 `NDAC timeout!GPIBbus::sendCmd: Failed to send command 3F to device 18`
What could be the reason for this?

I tried to send an "Instrument Preset" command but I have no idea what a normal error free debug log should look like.

Twilight-Logic, 2022-05-03:
Thanks. Another good spot! The quotes are missing around the character string. Change:

    `debugStream.println(F(id_h: done.));`

To:

    `debugStream.println(F("id_h: done."));`

I have now gone through and checked all the other debug statements. I found one more error which I have corrected. An update has been published to the GitHub.

With regards to the timeout, 3F is 'untalk' and would be the very fist byte sent in any command sequence. The error suggests that the HP8591A is not responding on the bus. There can be a number of reasons:

- problem GPIB cable
- problem with interface wiring
- GPIB interface not configured/enabled on the instrument
- instrument GPIB address not set using the ++addr command

If its none of the above, then if you have a logic analyser, you would be able to see what is actually being transmitted on the wire and each side of the SN7516x chips.

Mario-H, 2022-05-03:
Thanks for another quick fix!

As for my other problem:

- Address was set:
NDAC timeout!GPIBbus::sendCmd: Failed to send command 3F to **device 18**
- Changed cables
- There is no way to disable interface on the instrument
- checked the wiring multiple times (with conductivity tester) from µP to SN7516X chips and from there to the GBIP-connector.
- used ++xdiag to check all control and data lines
But I found no reliable way to really test the "out" side of the SN7516X chips. I assume this is because ++xdiag doesn't control TE and DE lines during the ++xdiag tests.

Since I don't own a logic analyser: 
Is there a location in the code where I could insert a delay (say 2 to 5  seconds) before the next data (or bit) is written to the SN7516X chips so I can monitor the individual lines sort of real time?

Twilight-Logic, 2022-05-03:
That is an interesting point regarding ++xdiag and the outside pins of the SN75161x. I could possibly add this feature. Will need to have a think about how to accomplish it. Setting digitalWrite(SN7516X_TE,HIGH); just before gpibBus.setDataVal(val); (line 2141) or gpibBus.setControlVal (line 2147) should probably do it. For the data bus (case 0), it will need to be set back to LOW again after the delay. For the control bus (case 1), this will automatically happen when the command in line 2153 or 2155 gets executed.

When using ++xdiag, the selected pin state should be active for around 10 seconds before reverting back to idle. However, the interface should not be connected to anything during testing.

Which pin are you using for TE at the moment? I was puzzled in your opening post as to why you would be using pin 7 as that is allocated to the ATN signal? You probably want to be sticking with pin 6 and pin 13 (or 3 [REN]) for DC. When I was testing, I originally drove DC from a separate pin but later discovered that it was sufficient to simply connect it to REN. The effect is the same.

Mario-H, 2022-05-03:
Would be great if you would be able to implement controlling TE and DC with ++xdiag for configs using SN7516X chips.

With the delay I mean a sort of slow execution of the commands when writing to de bus. e.g.:
- Lower NDAC	& serial print "lowering NDAC"
delay(x)
- Raise NRFD	& serial print "Raising NRFD"
delay(x)

etc.

That would create some time to use a scope to see if what is supposed to happen according to the serial log is in fact visible on the scope.

As for TE:
In the schematic on page 42 of manual (version 0.49.11) TE is connected to pin 6 and DC to pin 13
So that's what I built

in AR488_Layouts.h I found:
```

/**************************************/
/***** UNO/NANO LAYOUT DEFINITION *****/
/***** vvvvvvvvvvvvvvvvvvvvvvvvvv *****/
#if defined(AR488_UNO) || defined(AR488_NANO)

/***** NOTE: UNO/NANO pinout last updated 21/09/2019 *****/
#define DIO1  A0  /* GPIB 1  : PORTC bit 0 */
#define DIO2  A1  /* GPIB 2  : PORTC bit 1 */
#define DIO3  A2  /* GPIB 3  : PORTC bit 2 */
#define DIO4  A3  /* GPIB 4  : PORTC bit 3 */
#define DIO5  A4  /* GPIB 13 : PORTC bit 4 */
#define DIO6  A5  /* GPIB 14 : PORTC bit 5 */
#define DIO7   4  /* GPIB 15 : PORTD bit 4 */
#define DIO8   5  /* GPIB 16 : PORTD bit 5 */

#define IFC    8  /* GPIB 9  : PORTB bit 0 */
#define NDAC   9  /* GPIB 8  : PORTB bit 1 */
#define NRFD  10  /* GPIB 7  : PORTB bit 2 */
#define DAV   11  /* GPIB 6  : PORTB bit 3 */
#define EOI   12  /* GPIB 5  : PORTB bit 4 */

#define SRQ    2  /* GPIB 10 : PORTD bit 2 */
#define REN    3  /* GPIB 17 : PORTD bit 3 */
#define ATN    7  /* GPIB 11 : PORTD bit 7 */      <= conflicts with definition below
```
Indicating that pin 6 and 13 are free. So far so good.

But in config.h the relevant lines are:
```
//#define SN7516X
#ifdef SN7516X
  #define SN7516X_TE 7       //      <= conflicts with definition above
//  #define SN7516X_DC 13
//  #define SN7516X_SC 12
#endif
```
Suggesting to use pin 7 (but that pin is used for ATN as can be seen in the layout above)

That's why I changed the definition to:
```
#define SN7516X
#ifdef SN7516X
  #define SN7516X_TE 6   // changed to avoid conflict with ATN
  #define SN7516X_DC 13
//  #define SN7516X_SC 12
#endif
```
So the manual is in line with the schematic on page 42 but the code in the repository is not.

Twilight-Logic, 2022-05-05:
Thank you for that observation. The SN7516x support was developed on a Mega 2560 and a completely different set of GPIO pins were used for the GPIB port. The pin assigned to SN7516X_TE should be a spare GPIO pin. I can see how that would have been confusing. To make things consistent, I amend this to pin 6. The change should appear in the next update, which I am hoping will be made available in the next few days.

Regarding the delay, thank you for the clarification. I think I now understand what you mean. The xdiag function was implemented as a means to turn a GPIB signal wire on or off and hold it in that state for 10 seconds. The idea was to make it possible to change the state of one or any number of wires at a time and test that the correct GPIO pin was responding.

The place to insert such delays might be in the GPIBbus::readByte() and GPIBbus::writebyte() function. This is the state machine that handles reading or writing a byte to the GPIB bus. When reading, the interface controls NRFD and NDAC. When writing it controls DAV. The comments indicate when each goes up and down ('assert' means pull the signal LOW - when a signal is un-asserted, the signal is HIGH). bear in mind that with a delay of 2-5 seconds it is highly unlikely that you will get a handshake or response from any connected instrument. Most instruments measure the timeout period in tenths or hundreds of milliseconds. However, you should hopefully be able to at least observe NDAC, NRFD and DAV changing state at the start of the handshake.

Mario-H, 2022-05-05:
Thanks for the explanation.

I'm currently restoring an 30 year old laptop that I used in the nineties to control the very same analyzer I now try to control.
That laptop still has the NI-GBIP interface card in it but the power supply is dead (hopefully that's the only problem with it). When I receive the part and hopefully get the laptop working I can test if the analyzer still has a working interface.
When it does I know for sure the problem lies with the AR488 interface.

If I can't revive the Laptop I will follow the route of implementing the delays en hope that in the mean time you find the time to implement the fix for ++xdiag that also controls the TE and DC pins.

Mario-H, 2022-05-16:
So the restoration of the laptop was a no go.
In the attic I found another 30 year old desktop with ISA slots and a working power supply but no Disk.
Took me a while to gather a disk and install DOS but in the end the pc was operational.
After inserting the NI-GBIP interface card and installing the drivers I was able to connect my analyzer.
And I now can say my cable and analyzer are not the problem as they work perfectly on the old PC.

That tells me the problem is somewhere in my AR488 interface. I have a feeling that the software had some upgrades/modifications since the last test on an MEGA 2560 and some SN7516x related routines may not have been updated.
The last thing I can do to verify this is to remove the SN7516x chips and put wire jumpers in the sockets to bypass the chips.
If the interface does work with SN7516x disabled then we know for sure where the problem lies.

Mario-H, 2022-05-19:
Progress!

Today I tested the interface with wire bridges in the IC-sockets of the two SN7516X chips.
Deactivated #define SN7516X in AR488_Conig.h, recompiled and uploaded the sketch.
Result: The analyzer responds to the commands I issue on the serial monitor.
So now I know that everything before and after the SN7516X chips is working as intended.
Next is getting the SN7516X chips involved.

I'm not 100% sure what you mean by:

> substituting the macros in GPIBbus::setControls() with your actual pin numbers and removing the #ifdef SN7516x wrapper.

Should I remove #ifdef SN7516x in AR488_Conig.h or in GPIBbus::setControls() or in both?

Twilight-Logic, 2022-05-21:
Dont’t change AR488_Config.ini. Only edit the AR488_GPIBbus.cpp file. In that file you will find a function called GPIBbus::setControls. In that function you need to:

- comment out any #ifdef ... #else ...  #endif statements
- Comment out any lines referring to SN75161_DC and SN75161_SC since you have not configured them.
- in the remaining lines, replace any instance of SN75161_TE and SN75161_DC with the actual pin number you are using.

Unfortunately I am rather ill and laid up in bed otherwise I would have provided you with the modified function.

The idea is to bypass and not rely on the macro definitions but to set the pins ditectly. It is possible the substitutions are not ocurring automatically as they  I am hoping that will work and provide at least a temporary solution until I am well enough to research the problem more fully.

Mario-H, 2022-05-21:
Thanks for the clarification.

Hope you recover soon from you illness.

So I replaced all references to SN75161_TE and SN75161_DC in GPIBbus::setControls() with the actual pin numbers (6 and 13 in my case)
Left #define SN7516X in config.h enabled and the wire jumpers in their place.
Uploaded the sketch and all worked.
Next I removed the wire jumpers from the SN75160 socket and inserted the SN75160 chip.
Tested again and it worked.
So the SN75160 control does not seem to be the issue.

Next I removed the wire jumpers from the SN75161 socket and inserted the SN75161 chip.
Tested again but this time the analyzer did not respond to the issues commands.

Before your last message I was going through the code in GPIBbus::setControls and put the states of TE, DC and SC in a table.
What I noticed is that DC and SC are only switched in CINI and DINI cases.
In all other cases they are left in their previous states it seems.
Could that be part of the problem?

Any suggestions on how to proceed troubleshooting?

[image]
?

Twilight-Logic, 2022-05-21:
The DC signal controls ATN, REN and IFC. In controller mode we transmit those signals, in device mode we listen to them. Thats why they only need to be set once depending on whether we are in Controller or Device mode. By default they are set in controller mode, but are switched when the user changes mode.

TE on the other hand controls the direction of data flow and so gets set appropriately depending on whether we need to read or write.

I have had previous confirmation that the SN7516x support works, although I am not dismissing your report. However, could I trouble you to just re-check the wiring around the SN75161 chip. Since your analyzer responded with the SN75160 in place, this suggests that TE states are switching OK, at least to that chip. If you have set the correct pin number for DC then that shouln’t be an issue either.

Mario-H, 2022-05-21:
I re-checked the wiring from Pin 13 on the ATMEGA328 chip to the DC-pin on the SN75161 chip and from pin 6 on the ATMEGA328 chip to the DC-pin on the SN75161 chip:
No fault found.
To make sure I also checked Vcc and Gnd of the SN75161 chip: 
No fault found.

I did however find some references to SN7516X_DC and SN7516X_TE in AR488.ino and replaced them with the real pin numbers:
Still no response from the analyser.

I also removed the lines referring to the build-in led (my print doesn't have a led) in case that interfered with the DC signal on pin 13:
Still no response from the analyser.

Uploaded a simple blink sketch to toggle pin 13: Signal is present on DC-pin of SN75161.
Did the same for pin 6: Signal is present on TE-pin of SN75161 and Signal is present on TE-pin of SN75160.

Can you advise on any other tests I can perform?

Twilight-Logic, 2022-05-22:
There should be no neccesity to change anything in the main .ino file. I already have positive proof that those substitutions are working Ok by observing the output in the preproc directory which is found in the temp build directory created automatically by the IDE. I couldn’t see the same for the GPIBbus.cpp file though.

It looks like you have done some thorough testing and confirmed that the change of state on the GPIO pin is being reflected on the corresponding SN75161 pin. This unfortunately leaves the conclusion that the chip might be faulty? Do you have another chip to try?

When you send a command to the instrument, both REN and ATN should go from high to low. Are you seeing that on the output from the SN75161?

Mario-H, 2022-05-22:
I already came to the same conclusion that possibly the chip itself is the problem since my findings point less and less to the code being the problem.
Yesterday I ordered a replacement that should arrive Tuesday.

In the mean time I tested REN and ATN while sending a command as you suggested, but nothin happened on the output side of the SN75161
This again points at the chip itself being the cause.

When the chip arrives I will post an update of the results.

Twilight-Logic, 2022-05-22:
It is rather unusual for one of these to fail so I am still trying to keep an open mind. We will know for sure when the new chip arrives.

In the meantime, with the interface powered on and in standby (doing nothing) what states do you see on REN, IFC and ATN? I am guessing that „nothing happened” in your previous update meant that no voltage was present at any time?

I think you already checked VCC and please forgive the dumb question, but you are sure you have 0V on pin 10 AND it is connected to the common ground of the rest of the circuit (abscence of voltage is not sufficient) ? Apologies if you have already checked this.

Mario-H, 2022-05-23:
Correct, no voltage changes occurred at REN, IFC and ATN when I issued a command.
Yes I checked Vcc and Ground (measured against central ground and across Pin 10 and 20).

I'm not sure where I got the chips (had them for years) could be China, eBay or a more reliable source.
So I'm reluctant to dismiss the chip being the problem.

Hopefully we know more tomorrow.

Twilight-Logic, 2022-05-23:
I guess if you have 5V across pins 10 and 20 and pin 20 and „central ground” then the chip is grounded and must be getting power.

Since with GPIB a signal is inactive when HIGH, both IFC and ATN should be at 4.5-5V. When a command is issued ATN should get pulled LOW indicating that it is asserted. IFC gets pulled LOW for a few milliseconds during startup but should otherwise stay HIGH. REN would get pulled LOW at startup in controller mode or possibly when the first command is Either way you should have been able to see something happening on REN and ATN. Lets hope the new chip fixes it.

Mario-H, 2022-05-24:
The new chip arrived today.
Put it in the socket and tested the interface.
SUCCESS it finally works!

I feel I must apologize for taking up so much of your time helping me to figure out why my interface wasn't working.
Thanks a lot for all your support and patience with me and mij interface.

Now I can start with coding some software to control my analyzer.

Mario-H, 2022-05-24:
Thanks.
All test were done using the modified code.
I only disabled the debug functions so the output is only analyzer results.
I already downloaded the latest version of the software from GitHub but have not yet uploaded it to the interface.
As soon as I have done that I will report back if that works out of the box or not.

Mario-H, 2022-05-25:
So I uploaded the latest version (0.51.07) to the interface.
I only enabled #define SN7516X and #define SN7516X_DC 13 in AR488_Config.h
The analyzer responds to all the command I send and I can read the response with ++read.
So now we both know this version works as expected with an barebone ATMEGA328 and the two SN7561X chips
FYI: I use a FTDI USB-Serial interface and added a jumper to disable reboots when opening the com port.
When I need to update the software I only have to put the jumper in to enable DTR to reset the ATMEGA328.
Perfboard setup of my interface:

[image]

Twilight-Logic, 2022-05-25:
Thank you for taking the time and trouble to do that and confirm the outcome. I was worried that there might still be a problem to resolve in the code but I am releived it works OK a I don't have to worry about is and can concentrate on my recovery!

You have done a great job of the interface. Those wires are nicely and neatly arranged. I think this is the first one I have seen done on Perfboard. You don't see it so much of it  these days. I have used it occasionally myself. Congratulations on a job well done!

I think I can now close this issue.
Regards.

Mario-H, 2022-05-25:
Thanks for the compliments. Fore one time small projects I still use perfboards now and then.
Sure go ahead and put it in the manual. Perhaps other users can use it.

Next I'm going to look into detecting errors on the GPIB-bus.
I've been reading the 30 year old manuals from my National Instruments interface and there were several functions available for detecting errors.

Thanks for all you support and I hope your recovery will be a quick one.