Twilight-Logic/AR488 #27: ESP32 status

rhaamo, 2022-05-08:
ESP32 directly with wireless builtin would be very great, what is the status of being able to use a nodemcu or equivalent in pin count ?

Twilight-Logic, 2022-05-08:
dourada has ported the AR488 code to the ESP32. You can find his project here:

https://github.com/douardda/AR488

I am not sure whether he has used a 30 or 38 pin version of the NodeMCU board but considering the price difference is small, it is probably worth having the extra pins. I haven't tested his code myself, but hopefully he will be able to tell you more.

rhaamo, 2022-05-08:
oh great thanks!

maybe it would be better to only have one project for everything.

I will look at their fork and give it a try.

closed 2022-05-21 by Twilight-Logic