Twilight-Logic/AR488 #32: ++spoll compatibility problem

dg8saq, 2022-10-09:
Hi,

I think I have detected an incompatibility between the AR488 and the Prologix adapter that seems to cause an issue with the KE5FX VNA toolbox.
The issue is with the ++spoll command. The KE5FX software issues it with an address, i.e. ++spoll 16, where 16 is the primary GPIB address of the polled instrument.

Here is the command sequence that works successful with the Prologix adapter in combination with the HP8510B VNA:

++addr
  16
++spoll
  1
++spoll 16
  1

For better visibility I have indented the adapter responses. So far so good.

Now, here is the same sequence using the AR488:

++ver real
  AR488 GPIB controller, ver. 0.51.09, 20/06/2022
++addr
  16
++spoll
  1
++spoll 16
 => no response from the AR488 which causes the KE5FX VNA tool to lock up.

Have I misunderstood something or can there be done something about this problem?

Thanks and best regards,
Tom DG8SAQ

Twilight-Logic, 2022-10-10:
Thank you for the report. I will investigate.

Twilight-Logic, 2022-10-10:
I think I found a bug in the spoll_h function. Please try the attached and let me know whether it works. If it does then I will publish this as a release.

[AR488-test.zip](https://github.com/Twilight-Logic/AR488/files/9747602/AR488-test.zip)

Twilight-Logic, 2022-10-11:
Thank you for reporting back. I have pushed an update to the repository. I am not sure regarding your other issue. Will perhaps need to have a further look at that. Maybe the auto read loop is getting ahead of the instrument before it is ready to send the next reading.

closed 2025-04-09 by Twilight-Logic