Twilight-Logic/AR488 #39: Not enough digitalWrite's of SN7516X_DC?

RussellSenior, 2023-06-04:
There only seem to be a few instances of digitalWrite(SN7516X_DC, LOW or HIGH). And yet, according to the [SN75161 datasheet](https://www.ti.com/lit/gpn/sn75161b) on the second page's "SN75161B RECEIVE/TRANSMIT" function table, this is an important bit in setting the direction of the control lines through the chip to allow the appropriate handshaking. Furthermore, the pin directions set in the AR488_GPIBbus.cpp code seem inconsistent with the TE and DC levels set, according to the table.

RussellSenior, 2023-06-07:
Reading further, it seems that the SN7516X_DC is changing only on mode (controller vs device) changes, and that looks right. I am finding inconsistencies in input vs output control pins vs the function table when considering the state of DC and TE.

Twilight-Logic, 2023-06-11:
Russel, thank you for your comments.

Its a while since I looked at this but yes, DC gets switched on mode. Other switching gets done depending on whether data is being read or written to the bus or the controller being switched to idle state.

The datasheet does take a little fathoming, but if you have any specific examples of inconsistencies, then I will be happy to dig out my test rig and have a look.

RussellSenior, 2023-06-11:
See: https://github.com/Twilight-Logic/AR488/pull/38 for some pin direction inconsistencies and setting pins that should be inputs.

Twilight-Logic, 2023-12-21:
Russell,

Thanks for pointing that out. I initially merged your proposed changes without testing them, but then found that I couldn't talk to my HP34401A so I ended up reverting them. However, I did like the idea and spent some time thinking about it and adding in your changes a step at a time. As I started to analyse things and work through each section of code, it became evident that there were some bit switching inconsistencies as you point out. Your comments in the code were very helpful.
One thing I will mention is that the reason for setting input pins to 1 (HIGH) is to enable the internal pull-up resistors which is essential for proper operation. The inputs don't want to be set to 0 (LOW) which would put them in high-z mode and could leave them floating. I agree it does look a bit strange though at first sight. This is at least true for Arduino boards, but I am not sure about other boards such as the ESP32.
I have attached an updated version of the code which implements your idea, although the way I have implemented it is slightly different. I had the thought to separate functions according to the two distinct signal groups which are the a) control signals group (IFC, IFC, REN, SRQ) and b) the handshaking signals group (DAV,NDAC,NRFD,EOI). The former gets changed infrequently - usually when the operating mode of the interface is changed from controller to device and back, whereas the latter group changes with every character sent. If figured this added a bit more clarity. The operation is pretty much the same and hopefully achieves the same goal, i.e making the code easier to read and understand.
I would appreciate it if you have the opportunity to test the attached and let me have your feedback.
[AR488-0.51.25-test.zip](https://github.com/Twilight-Logic/AR488/files/13742109/AR488-0.51.25-test.zip)

Twilight-Logic, 2024-02-14:
I have now pushed up the latest revision of the code to the GitHub. This includes corrections for the inconsistencies you mentioned in your pull request.

closed 2024-02-29 by Twilight-Logic