Twilight-Logic/AR488 #40: Untalk / Unlisten flag clearing code

afrb2, 2023-09-03:
This code handling UNT and UNL seems to set its flags in a strange way

AR488.ino line 2382

 if (ustat & 0x01) {
    if (!device_unl_h()) ustat &= 0x01; // Clears bit if UNL was not required
  }

  /***** Try to untalk bus *****/
  if (ustat & 0x01) {  // ? 0x2 
    if (!device_unt_h()) ustat &= 0x02; // Clears bit if UNT was not required
  }

Second one would appear should if(ustat & 0x2) and to clear the bit maybe an XOR rather than &= ?  Don't think the bit clearing is very important though!

Twilight-Logic, 2023-09-05:
Thank you for spotting that. You are correct. It should be checking the second bit and should read:
```
if (ustat & 02) {
    if (!device_unt_h()) ustat &= 0x02; // Clears bit if UNT was not required
}
```
I will fix it in the next release.

afrb2, 2023-09-05:
On Tue, 5 Sep 2023 at 08:08, Twilight-Logic ***@***.***>
wrote:

> Thank you for spotting that. You are correct. It should be checking the
> second bit and should read:
>
> if (ustat & 02) {
>     if (!device_unt_h()) ustat &= 0x02; // Clears bit if UNT was not required
> }
>
> Also -- is the &= 0x2 right?  I’d expect &= ~0x2  to mask out the bit.  Or
maybe stat ^= 0x2 since we know it was set.

Alan

> Message ID: ***@***.***>
>

Twilight-Logic, 2023-11-08:
Sorry, yes, my typo. That should have bveen '0x2' and not '02'.

You are also correct that it should be ustat &= ~02, or else, as you point out, we can simply toggle the bit with ^= 0x2 because we already know that it is set.

I will make the necessary changes in the code.
Thanks.

Twilight-Logic, 2024-05-10:
This should be fixed in version 0.51.29

closed 2025-04-08 by Twilight-Logic