Twilight-Logic/AR488 #41: EOI signal mystery?

m0pub, 2023-11-04:
I'm sorry if this issue/question has already been posted, but I have been using my AR488 with an old Tektronix 'scope. With default AR488 settings it works well for sending and receiving 'standard' messages, such as *idn?.
However, I wanted to receive binary data which contains LF and CR characters, so I configured my AR488 to use EOI to signal the end of each message with "++eoi 1" and "++eos 3". But this stopped all communication from working.
Clearly the issue was with the EOI signal, and a quick look at the AR488 source code showed that the AR488 asserts EOI (for 40us) _after_ the last character of a message has been read and acknowledged by the device. I'm not an IEEE488 expert by any means, but I think that the standard says EOI will be asserted _during_ the last character of the message, rather than afterwards. The really strange thing is that the AR488 firmware has clearly been written to function as I would expect, but then modified to behave differently.
As soon as I changed the firmware to assert EOI during the last character of each message, everything starting working. Has anyone else encountered this, and is it a bug or a feature, please?
Best wishes to you all, Alan.

Twilight-Logic, 2023-11-07:
A;lan, thanks for reporting this. The EOI signal should be asserted during the last character. I will investigate.

m0pub, 2023-11-07:
Thank you very much for your response. Please don't hesitate to ask me if you need any more details.
Best wishes, Alan

Twilight-Logic, 2023-11-07:
Alan, I would be interested to know which versions you compared, and what change you made?
Also, which Tektronix scope were you working with?

m0pub, 2023-11-08:
I am using version 0.51.18, dated 26/02/2023. I have not compared with earlier versions, but GPIBbus::writeByte() takes a second parameter to indicate whether EOI should be asserted during that byte, and in the current firmware it is **always** set to 'NO_EOI'.

My changes were very simple:
1) In GPIBbus::sendData() I changed:
err = writeByte(data[i], NO_EOI);
to
err = writeByte(data[i], (i == (dsize-1)));
so that EOI is asserted during the last character of the message - see AR488_GPIBbus.cpp line 627
2) Also in GPIBbus::sendData() I removed the code to assert EOI for 40us after the last byte is sent - AR488_GPIBbus.cpp lines 666 to 674.

My 'scope is a Tektronix TDS640, and I needed to send binary data to it in order to backup the NVRAM contents. Incidentally, I note that I originally referred to _receiving_ binary data, but that was very ambiguous. I meant that the 'scope needs to receive binary data, and hence the AR488 needs to send it - sincere apologies for any confusion.

Twilight-Logic, 2023-11-08:
Thanks for the additional info. I need to investigate this further, but your changes make sense. My only concern is what happens if CR, LF or CRLF is also enabled. The EOI would need to be sent on either the CR or LF accordingly, so modify it further to make sure that EOI is always asserted on the last character regardless of the type of data being sent and the settings being used.

If I send you an update, would you be able to test it on you TDS640 please? I want to be sure that any update I release doesn't break anything. I can test receiving, but I don't have an instrument I can test sending data to.

m0pub, 2023-11-08:
Yes indeed, a similar change needs to be applied to the cases where a CR, LF or CR/LF message terminator is enabled. I haven't attempted that since it wasn't a requirement for me, but of course it is in the more general case.
I would be happy to test changes on my TDS640, although I do not know how it will respond if I add the extra message termination characters.
(Incidentally, my AR488 interface uses an ESP32 CPU. This is not relevant to the EOI issue, which is independent of CPU type, but I did need to make one or two other little changes before the firmware would even compile for the ESP32. Would you like me to open a separate issue for that?)

Twilight-Logic, 2023-12-21:
Alan,

After a couple of weeks of ilness and other issues I have finally found some time to have a look at this now that we are nearing the holiday period. I have attached the version I am working on which inludes you fix and some additional code to handle CR/LF if it happens to be used. Are you able to test this on your TDS640 please?

With regards to the ESP32, it has been on my todo list add this board to the list of those supported, so I am very much interested in the changes that you had to make. There is also already a forked ESP32 version available here as well:

https://sr.ht/~douardda/AR488-ESP/

This version has some additional functions which I am hoping to backport into my project. I was also intending to add some WiFi functions and will be enabled if an ESP32 board is detected.

Here is the current version of the code. There have been some substantial changes including the one requested by yourself so I would be most interested in your feedback.
[AR488-0.51.25-test.zip](https://github.com/Twilight-Logic/AR488/files/13741978/AR488-0.51.25-test.zip)

m0pub, 2023-12-22:
Thank you for your new version of the AR488 firmware. I'm looking forward to trying it, but it may be after Christmas before I have any news. I'm sorry to hear that you have been ill, but very pleased that you are better now.
Thanks again, and best wishes, Alan.

Twilight-Logic, 2023-12-24:
Alan, thank you for your feedback.

My attempt to compie fr the ESP32 actually came up with the same problems. I started by creating a custom profile as you did.

For item 1 I had to make the same edit.

For item 2 I just defined oct:

 ` char oct[4] = {'\0'};`

However, since the value of cnt has already served its purpose having been sent to output, the variable can indeed be re-used. 

For item 3,  I also added an #ifdef clause for the ESP32 to include ESP.restart(). I note however the need to include the esp_task_wdt.h header.

I also ran into a problem with  the snprintf statement in the void id_h function in AR488.ino, line 2180. This contains a reference t the 'lu' formatting token which expects a long int variable. This had not surfaced previously when compiling for Arduino boards so I it would seem that the Uint-32 variable gpibBus.cfg.serial is equivalent to a an unsigned long int on an Ardiuino, but apparently not on an ESP32 despite this also supporting 32-bit values. I solved this by first assigning its value to an unsigned long int variable within the if statement that contains the snprintf command, and then changing the snprintf statement to refer to this new variable:

```
      if (strncasecmp(keyword, "serial", 6)==0) {
        unsigned long int serialnum = gpibBus.cfg.serial;
        memset(serialStr, '\0', 10);
        snprintf(serialStr, 10, "%09lu", serialnum);  // Max str length = 10-1 i.e 9 digits + null terminator 
        dataPort.println(serialStr);
        return;    
      }
```
After that it compiled without error or warning.

Thank you for taking the time to test with your TDS640.  I was able to test reading my DMMs, but not sending data from an oscillocope which involves sending much more data and using different functions, so it is good to know that this works as well.

m0pub, 2023-12-27:
Interesting - I didn't have any problem with the "%lu" format token when building for the ESP32. I have just checked, and your original code works fine for me without any change. This could be something to do with compiler options, but unfortunately I don't have time to investigate at the moment.
Best wishes, Alan

Twilight-Logic, 2024-02-14:
I have finally managed to push up the latest update which fixes the EOI issue.

The "%lu" issue seems to be down to the fact that a different compiler is used for some boards, e.g. ESP boards. It seemed to be fine for some boards and not others. I was able to find a common option that would work for any board I tried compiling for.

closed 2024-02-29 by Twilight-Logic