Twilight-Logic/AR488 #42: Not compile more errors

selevo, 2023-11-27:
AR488.ino: In function 'void setup()':
AR488.ino:383:13: error: 'flushPbuf' was not declared in this scope
AR488.ino: In function 'void loop()':
AR488.ino:568:24: error: 'execCmd' was not declared in this scope
AR488.ino:575:35: error: 'sendToInstrument' was not declared in this scope
AR488.ino:620:15: error: 'tonMode' was not declared in this scope
AR488.ino:622:15: error: 'lonMode' was not declared in this scope
AR488.ino:625:20: error: 'attnRequired' was not declared in this scope
AR488.ino:631:33: error: 'flushPbuf' was not declared in this scope
AR488.ino:659:48: error: 'serialIn_h' was not declared in this scope
AR488.ino: In function 'uint8_t serialIn_h()':
AR488.ino:702:46: error: 'parseInput' was not declared in this scope
AR488.ino: In function 'uint8_t parseInput(char)':
AR488.ino:746:20: error: 'addPbuf' was not declared in this scope
AR488.ino:752:23: error: 'flushPbuf' was not declared in this scope
AR488.ino:755:26: error: 'showPrompt' was not declared in this scope
AR488.ino:764:38: error: 'isCmd' was not declared in this scope
AR488.ino:768:27: error: 'flushPbuf' was not declared in this scope
AR488.ino:774:70: error: 'isIdnQuery' was not declared in this scope
AR488.ino:776:25: error: 'flushPbuf' was not declared in this scope
AR488.ino:793:20: error: 'addPbuf' was not declared in this scope
AR488.ino:805:18: error: 'addPbuf' was not declared in this scope
AR488.ino:817:19: error: 'isCmd' was not declared in this scope
AR488.ino:821:17: error: 'flushPbuf' was not declared in this scope
AR488.ino: At global scope:
AR488.ino:904:23: error: 'addr_h' was not declared in this scope
AR488.ino:905:40: error: 'aspoll_h' was not declared in this scope
AR488.ino:906:23: error: 'amode_h' was not declared in this scope
AR488.ino:907:40: error: 'clr_h' was not declared in this scope
AR488.ino:908:40: error: 'dcl_h' was not declared in this scope
AR488.ino:909:40: error: 'default_h' was not declared in this scope
AR488.ino:910:23: error: 'eoi_h' was not declared in this scope
AR488.ino:911:23: error: 'eor_h' was not declared in this scope
AR488.ino:912:23: error: 'eos_h' was not declared in this scope
AR488.ino:913:23: error: 'eot_char_h' was not declared in this scope
AR488.ino:914:23: error: 'eot_en_h' was not declared in this scope
AR488.ino:915:23: error: 'help_h' was not declared in this scope
AR488.ino:916:40: error: 'ifc_h' was not declared in this scope
AR488.ino:917:23: error: 'id_h' was not declared in this scope
AR488.ino:918:23: error: 'idn_h' was not declared in this scope
AR488.ino:919:23: error: 'llo_h' was not declared in this scope
AR488.ino:920:23: error: 'loc_h' was not declared in this scope
AR488.ino:921:23: error: 'lon_h' was not declared in this scope
AR488.ino:922:23: error: 'macro_h' was not declared in this scope
AR488.ino:923:40: error: 'sendmla_h' was not declared in this scope
AR488.ino:924:23: error: 'cmode_h' was not declared in this scope
AR488.ino:925:23: error: 'sendmsa_h' was not declared in this scope
AR488.ino:926:40: error: 'sendmta_h' was not declared in this scope
AR488.ino:927:40: error: 'ppoll_h' was not declared in this scope
AR488.ino:928:23: error: 'prom_h' was not declared in this scope
AR488.ino:929:23: error: 'read_h' was not declared in this scope
AR488.ino:930:23: error: 'rtmo_h' was not declared in this scope
AR488.ino:931:23: error: 'ren_h' was not declared in this scope
AR488.ino:932:23: error: 'repeat_h' was not declared in this scope
AR488.ino:933:40: error: 'rst_h' was not declared in this scope
AR488.ino:934:23: error: 'trg_h' was not declared in this scope
AR488.ino:935:40: error: 'save_h' was not declared in this scope
AR488.ino:936:23: error: 'setvstr_h' was not declared in this scope
AR488.ino:937:23: error: 'spoll_h' was not declared in this scope
AR488.ino:938:40: error: 'srq_h' was not declared in this scope
AR488.ino:939:23: error: 'srqa_h' was not declared in this scope
AR488.ino:940:23: error: 'stat_h' was not declared in this scope
AR488.ino:941:23: error: 'ton_h' was not declared in this scope
AR488.ino:942:40: error: 'unlisten_h' was not declared in this scope
AR488.ino:943:40: error: 'untalk_h' was not declared in this scope
AR488.ino:944:23: error: 'ver_h' was not declared in this scope
AR488.ino:945:40: error: 'verb_h' was not declared in this scope
AR488.ino:946:23: error: 'xdiag_h' was not declared in this scope
AR488.ino: In function 'void execCmd(char*, uint8_t)':
AR488.ino:1036:15: error: 'getCmd' was not declared in this scope
AR488.ino: In function 'void setvstr_h(char*)':
AR488.ino:1958:16: error: 'id_h' was not declared in this scope
AR488.ino: In function 'void attnRequired()':
AR488.ino:2450:21: error: 'device_listen_h' was not declared in this scope
AR488.ino:2457:23: error: 'device_unl_h' was not declared in this scope
AR488.ino:2462:23: error: 'device_unt_h' was not declared in this scope
AR488.ino:2521:26: error: 'execGpibCmd' was not declared in this scope
AR488.ino:2548:23: error: 'device_listen_h' was not declared in this scope
AR488.ino:2560:21: error: 'device_talk_h' was not declared in this scope
AR488.ino: In function 'void execGpibCmd(uint8_t)':
AR488.ino:2615:22: error: 'device_spe_h' was not declared in this scope
AR488.ino:2622:22: error: 'device_spd_h' was not declared in this scope
AR488.ino:2626:22: error: 'device_unl_h' was not declared in this scope
AR488.ino:2630:22: error: 'device_unt_h' was not declared in this scope
AR488.ino:2634:22: error: 'device_sdc_h' was not declared in this scope
Return code is not 0

Twilight-Logic, 2023-11-27:
Could you show your AR488_Config.h file please?

selevo, 2023-11-27:
all files origunal from this git
not edit
 my be problem only with arduinodroud program
 In arduino ide online - normal build
https://youtu.be/HCp-8tU8e9Y?feature=shared

Twilight-Logic, 2023-11-28:
Thank you for confirming. Will look into it.

Twilight-Logic, 2024-05-10:
Just to confirm, ArduinoDroid does not yet have full compatibility with all Arduino projects written in the Arduino IDE.

closed 2024-05-10 by Twilight-Logic