Twilight-Logic/AR488 #44: Device mode custom commands etc.

FKW9, 2024-03-12:
I don't quite understand the current state of device mode.

Is it possible to use the arduino as a device with a specific IEEE488 address, so i can communicate with it without the need of USB?
For example making write/read/query operations to the device over GPIB to e.g. return the state of a I/O?

Twilight-Logic, 2024-03-12:
May I ask what you would like to achieve? If the question is what  think it is, then you might be referring to secondary commands. The AR488 project is primarily designed to facilitate a USB connection to an older device with a GPIB interface. It therefore implements the  primary GPIB commands that are needed to, e.g. read measurements or remote control the device.

If you are looking for capability to run secondary addressing commands such as reading and writing files from a disk drive, then it is necessary to understand which device is to be interacted with. Although there are some common features, each vendor and device has its own implementation of a command set. The following project, is based on the AR488 code, emulates a Tektronix tape drive. The project adds an SD card reader to the Arduino and its main function is to allow the Tektronix computer to create/erase/read/write files to the SD card over the GPIB bus. In a similar way, if one has details of the command set, it ought to be possible to emulate other devices. For example, I had considered implementing an emulation of a Tektronix and a Commodore PET disk drive although, for the latter, emulator options already exist. The USB cable is used mainly for programming the Arduino, but also a handful of test commands which do return responses over serial, but other than that, no data or device commands are sent over serial. However, it could just as easily contain the full AR488 command set.

https://github.com/Twilight-Logic/Tek405xEmulator

So yet, it is possible to set the AR488 with a specific IEE488 address and communicate with it over the GPIB bus from another GPIB node or controller. However, in order to answer your question in more specifically, I would need to understand better what you are looking for.

FKW9, 2024-03-12:
Thanks for your detailed answer.

I'm using the Arduino MEGA with some external circuitry on a custom made PCB.
Since every instrument in my laboratory is connected via GPIB, I also wanted to control this custom made device with it so I don't have to change my software routines.

Therefore, I can't emulate a device, but I am building an entirely new device that could theoretically implement any command set I'd like (in addition to the standard GPIB interface functions). 

The basic idea is that I connect it to my GPIB bus, send a query like `GPIOSTATe 10?` to the device address and get a response with the state of Arduino pin 10, e.g. `STATe LOW`. These custom commands must of course be implemented in the code.

I hope this makes my question a little clearer.

Twilight-Logic, 2024-03-13:
Thank you for the clarification. That's a bit like sending SCPI commands to an instrument. Yes, with a little bit of work it could be made to work. At the moment, in the AR488 code, data is just sent to the serial port because there is no defined device function. However, there is no reason why each received line could not, instead, be parsed for commands and parameters which then trigger routines to do something with GPIO pins and, if required, returning a result back to the requester via the GPIB bus. It would require writing a module that can be enabled in the config, one having a basic framework provided for someone to then add their own custom routines into. Its an interesting idea and I will have a think about how it could be implemented.

closed 2024-05-10 by Twilight-Logic

FKW9, 2024-05-23:
I tried to understand the source code but its pretty hard for me to implement/understand this functionality.
Controller mode works perfectly fine, but using device mode in listen-only and promiscuous mode raised some questions:

### ++prom mode
In ++prom mode, I do not see any data that is sent over the GPIB bus. For example, when I IDN a instrument on my bus on address 7 with my PC (NI PCIe-GPIB Controller), I get no output on the data port, only on the debug port:
```
AR488\AR488_GPIBbus.cpp:539 (receiveData) > Start listen ->
AR488\AR488_GPIBbus.cpp:540 (receiveData) > Before loop flags:
AR488\AR488_GPIBbus.cpp:541 (receiveData) > TRNb: 0
AR488\AR488_GPIBbus.cpp:542 (receiveData) > rEOI: 1

AR488\AR488_GPIBbus.cpp:599 (receiveData) > After loop flags:
AR488\AR488_GPIBbus.cpp:602 (receiveData) > Bytes read:  0
AR488\AR488_GPIBbus.cpp:603 (receiveData) > <- End listen.
AR488\AR488_GPIBbus.cpp:890 (setControls) > Set GPIB lines to idle state
AR488\AR488_GPIBbus.cpp:645 (receiveData) > done.
AR488\AR488_GPIBbus.cpp:877 (setControls) > Initialised GPIB listener mode
AR488\AR488.ino:541 (loop) > <- ATN loop finished
```
So it seems that in prom mode, the data cannot be received correctly?

### ++lon mode
In ++lon mode I see everything that is happening on the bus, despite what is written in the manual.
"_[...] lon mode works only with a Talk-Only node and where no controller is present on the GPIB bus, whereas prom mode can operate when a controller is present and addressing is used_"
This is weird because addressing **is** used and there **is** also a controller on the bus. Here is an example output from the AR488 data port while making a query operation with the GPIB Controller to a instrument with address 7:

QUERY:
ASCII: `@?'*IDN?\n`
HEX: `40 3f 27 2a 49 44 4e 3f 0a`

RESPONSE:
ASCII: `? Gmaturo gmbh,MCU,5170816,1.30\n`
HEX: `3f 20 47 6d 61 74 75 72 6f 20 67 6d 62 68 2c 4d 43 55 2c 35 31 37 30 38 31 36 2c 31 2e 33 30 0a`

I guess the first bytes are somehow related to addressing?
This looks more like a prom mode than a lon mode...?
___
Furthermore I tried to send data directly to the AR488 device (address 17) with my GPIB Controller.
Of course I will get a timeout, but the data from a query/write operation is received _somewhat_ correctly(?).
This is the serial output from the AR488 from a query and write operation in listen-only mode. Note that I do not get any debug information when using ++lon mode, despite having all debug defines uncommented.

ASCII: `@?1*IDN?\n? Q`
HEX: `40 3F 31 2A 49 44 4E 3F 0A 3F 20 51`

ASCII: `@?1TEST123\n`
HEX: `40 3F 31 54 45 53 54 31 32 33 0A`
___
When doing the same in ++prom mode, I get no serial output on the data port, but on the debug port I get the following:

QUERY: `*IDN?\n`

DEBUG OUTPUT ON AR488:
```
AR488\AR488.ino:539 (loop) > -> Attention signal detected
AR488\AR488_GPIBbus.cpp:901 (setControls) > Set GPIB lines to idle state
AR488\AR488_GPIBbus.cpp:901 (setControls) > Set GPIB lines to idle state
AR488\AR488_GPIBbus.cpp:539 (receiveData) > Start listen ->
AR488\AR488_GPIBbus.cpp:540 (receiveData) > Before loop flags:
AR488\AR488_GPIBbus.cpp:541 (receiveData) > TRNb: 0
AR488\AR488_GPIBbus.cpp:542 (receiveData) > rEOI: 1
<\r><\n>
AR488\AR488_GPIBbus.cpp:599 (receiveData) > After loop flags:
AR488\AR488_GPIBbus.cpp:602 (receiveData) > Bytes read:  0
AR488\AR488_GPIBbus.cpp:603 (receiveData) > <- End listen.
AR488\AR488_GPIBbus.cpp:890 (setControls) > Set GPIB lines to idle state
AR488\AR488_GPIBbus.cpp:645 (receiveData) > done.
AR488\AR488_GPIBbus.cpp:877 (setControls) > Initialised GPIB listener mode
AR488\AR488.ino:541 (loop) > <- ATN loop finished<\r><\n>
```

ANSWER ON PC:
```
Read Operation Error
VISA:  (Hex 0xBFFF0015) Timeout has expired before the operation is completed.
```

SCREENSHOT:
[image]

Again, in prom mode there seems to be an issue.
___
So, my current idea is to modify lon mode or implement a similar method:

1. check if the address matches our device address (set with ++addr cmd)
2. if the address is correct, parse the incoming data
3. if it is a query, prepare the bus and respond to the controller (_no clue how to do this yet_)

Have you thought about this idea any more?