Twilight-Logic/AR488 #45: Escaped characters not being sent

Mike0Github, 2024-05-09:
I am getting incorrect characters when trying to read HP3478A calibration memory at locations 10 and 27.

I am writing Wxx\n (87,xx,10) to the AR488 where xx is the memory location from 0 to 255, for escaped characters I am writing (87,27,xx,10), but if xx is 10, 13, or 27 nothing is written to the bus.

Monitoring the GPIB bus with another AR488 it appears the numbers 10, 13 and 27 are not being sent on the bus even though they are escaped with 27 (0x1B).  Other numbers escaped with 27 get through to the bus.

Is this a firmware bug or am I doing something wrong?

Twilight-Logic, 2024-05-09:
Mike, thank you for your report. Which version of the firmware are you using?

Could you please confirm which version of the firmware you are using please? There was a problem with this is previous version but it was fixed in the latest version.

Mike0Github, 2024-05-09:
I'm using version 0.51.28

Mike0Github, 2024-05-09:
I will install version 0.51.29 tomorrow and let you know.

Mike0Github, 2024-05-09:
Tried it this morning.  Works perfectly now.
b.t.w. During earlier testing I found I had to set the HP3478A to "Stop Trigger" (send 'T4') else it would send voltage readings along with the calibration data.
Thanks for a great program.

Twilight-Logic, 2024-05-10:
Thank you for the feedback. Glad it is working now.

closed 2024-05-10 by Twilight-Logic