Twilight-Logic/AR488 #50: AR488 at HP8753C: does not read plot data

Yrrah53, 2025-02-24:
I have three GPIB-USB interfaces: a genuine Prologix adapter, and two AR488 adapters, one with Bluetooth and and one with just USB. The AR488 adapters merely consist of a Pro Micro and a Centronics connector. The BT device has a HC-05 unit attached. SW version: 1.8.15/0.51.29. All adapters work fine with a Python script and a HP437B power meter. Goal is to get plot data from network analyzers: the HP8753C and a HP8756A with the K5FX  toolkit. The Prologix adapter works fine. The AR488 adapters have an issue with the HP8753C: upon the plot command only 11 bytes are transmitted (iso about 7700). Getting a plot out of a HP8563E spectrum analyzer this way works fine. I tested the AR488-HP8753C combi with some GPIB commands (like PRES, STAR, STB? etc): works fine. The interface was also monitored using the very useful Redirector tool, kindly provided by an expert who also gave me a lot of assistance. No luck so far with the HP8753C: plot data transmission is aborted after 11 bytes. What can be the issue?
Thanks in advance......

Twilight-Logic, 2025-02-25:
Thank you for your report and the details provided. Could I just confirm which version of the AR488 software you are using?
Also, could you send me a couple of separate files containing both the good and bad data from the HP8573C please?

Yrrah53, 2025-02-25:
Thanks for your respons.
Here are some files that maybe shed some light. These are actually
screenshots made with Redirector by DG8SAQ.
First file: tst_HP8753C..... Line 43/44/48 are interesting: the plot is
aborted after  some (11 as reported by 7470 emulator) bytes. The version is
in line 35.
A different situation gives similar output. File: 63E_LO.... This file is
recorded in a test set-up with a HP8563E and a HP437B power meter to
measure the LO output power of the sa. The SA is controller under sw
command of a TAM (Test and Adjustment Module). Text data are no problem at
all (ref lines 11-20). Then automatically a plot is send thereafter (line
33) and quickly aborted. This was repeated once with same result....
So the issue is: why is text data no problem and is plot data aborted? Same
happens when GPIB Configurator is used in conjunction with 7470.exe.
Best regards,
Y

Twilight-Logic, 2025-02-25:
Thank you. I don't see the attached files however? Just edit your post on Github and drag and drop the files into it please.

Twilight-Logic, 2025-03-09:
Could you attach a complete plot output using the Prologix please? I need to see the complete plot data.

Can you also tell me what settings you have for the following:

++eoi
++eos
++eor

Thank you.

Yrrah53, 2025-03-10:
Hello,

Thanks for your response. I think I already send it to you. Please refer to
annex (rtf format).
At line 16 you see that the regular Prologix adapter is used. Line 40 shows
the complete plot.
At line 72 you see I switched adapter to the AR488, nicknamed Prologix
6.101.
Line 81 shows the plot as it is received, only a few characters.
Thank you for your attention.  If you need more info/data, please let me
know.
Kind regards,
Harke

Twilight-Logic, 2025-03-11:
There is no line 72 or 81 in the files you have attached. File [tst_HP8753C_AR488_V3962.docx (renamed to .rtf) seems to end at line 50. I did try saving to a .txt file but it made no difference.

Is file 63E_LO_ampl_AR488.docx the actual plot?

Unfortunately encapsulating in another format such as RTF or DOCX mangles the data. I suspect the top part of the file is missing as I see no output from the Prologix, but I see clearly that the output from the AR488 is only ';;;;DF;IM;' and that you have set ++auto 1. The other parameters I asked for above will presumably be set to defaults.

Ideally it needs to be saved as a raw .txt or .bin file exactly as it comes out of the GPIB adapter.

In the meantime, I am wondering whether some character or other is being mis-interpreted. I would suggest trying setting the line terminator for the incoming data to either NL (++eor 2) or the EOI signal (++eor 7) and see whether that makes a difference.

Yrrah53, 2025-03-11:
I am very sorry, I am afraid I attached the wrong file? Attached is the
correct file. Saved as txt file. Now I hope the line numbers make sense. I
open/save the file in Notepad++. The first part is a recording of the data
transfer via Prologix, the second part with an AR488, nicknamed Prologix
6.101.

Best regards,
Harke

{\rtf1\ansi\ansicpg1252\deff0\deflang1043{\fonttbl{\f0\fnil Tahoma;}{\f1\fnil\fcharset0 Tahoma;}}
{\colortbl ;\red255\green0\blue0;\red0\green0\blue255;\red0\green0\blue0;}
\viewkind4\uc1\pard\cf1\f0\fs20 ++auto 1
\par ++eos 0
\par ++mode 1
\par ++addr 16
\par ++eoi 1
\par OUTPIDEN;
\par \cf2 HEWLETT PACKARD,8753C,0,4.13
\par \cf3 ComPort A disconnected
\par Connecting...
\par Connected!
\par Connected to client 127.0.0.1 , port = 8488
\par \cf1 ++eot_enable 0
\par ++ver
\par \cf2 Prologix GPIB-USB Controller version 5.3
\par \cf1 ++auto
\par \cf2 1
\par \cf1 ++eos 0
\par ++mode 1
\par ++addr 16
\par ++eoi 1
\par OUTPERRO
\par \cf2     0 ,"NO ERRORS"
\par \cf1 ++auto 0
\par ++loc
\par \cf3 Client 127.0.0.1 disconnected
\par Connected to client 127.0.0.1 , port = 8488
\par \cf1 ++eot_enable 0
\par ++ver
\par \cf2 Prologix GPIB-USB Controller version 5.3
\par \cf1 ++auto
\par \cf2 0
\par \cf1 ++auto 1
\par ++eos 0
\par ++mode 1
\par ++addr 16
\par ++eoi 1
\par OUTPPLOT;
\par \cf2\'03;;;;DF;IM;IP250,279,10250,7479;SC0 ,4095 ,0 ,4212 ;;PU;VS;SP1;;PU;PA0,0;PA244 ,1110 ;PD;PA244 ,1110 ;PA244 ,1110 ;PA260 ,1126 ;PA276 ,1116 ;PA292 ,1114 ;PA308 ,1112 ;PA324 ,1112 ;PA340 ,1112 ;PA356 ,1112 ;PA372 ,1114 ;PA388 ,1116 ;PA404 ,1118 ;PA420 ,1120 ;PA436 ,1120 ;PA452 ,1122 ;PA468 ,1122 ;PA484 ,1122 ;PA500 ,1122 ;PA516 ,1122 ;PA532 ,1122 ;PA548 ,1120 ;PA564 ,1120 ;PA580 ,1120 ;PA596 ,1118 ;PA612 ,1118 ;PA628 ,1116 ;PA644 ,1116 ;PA660 ,1114 ;PA676 ,1114 ;PA692 ,1112 ;PA708 ,1110 ;PA724 ,1110 ;PA740 ,1108 ;PA756 ,1106 ;PA772 ,1102 ;PA788 ,1098 ;PA804 ,1096 ;PA820 ,1092 ;PA836 ,1090 ;PA852 ,1084 ;PA868 ,1082 ;PA884 ,1080 ;PA900 ,1076 ;PA916 ,1072 ;PA932 ,1068 ;PA948 ,1066 ;PA964 ,1064 ;PA980 ,1060 ;PA996 ,1056 ;PA1012 ,1054 ;PA1028 ,1054 ;PA1044 ,1050 ;PA1060 ,1050 ;PA1076 ,1048 ;PA1092 ,1050 ;PA1108 ,1050 ;PA1124 ,1048 ;PA1140 ,1048 ;PA1156 ,1048 ;PA1172 ,1048 ;PA1188 ,1046 ;PA1204 ,1044 ;PA1220 ,1044 ;PA1236 ,1044 ;PA1252 ,1042 ;PA1268 ,1044 ;PA1284 ,1044 ;PA1300 ,1044 ;PA1316 ,1046 ;PA1332 ,1048 ;PA1348 ,1048 ;PA1364 ,1050 ;PA1380 ,1050 ;PA1396 ,1054 ;PA1412 ,1056 ;PA1428 ,1056 ;PA1444 ,1060 ;PA1460 ,1062 ;PA1476 ,1068 ;PA1492 ,1074 ;PA1508 ,1080 ;PA1524 ,1084 ;PA1540 ,1092 ;PA1556 ,1096 ;PA1572 ,1100 ;PA1588 ,1106 ;PA1604 ,1106 ;PA1620 ,1110 ;PA1636 ,1114 ;PA1652 ,1116 ;PA1668 ,1118 ;PA1684 ,1118 ;PA1700 ,1122 ;PA1716 ,1124 ;PA1732 ,1124 ;PA1748 ,1126 ;PA1764 ,1126 ;PA1780 ,1128 ;PA1796 ,1126 ;PA1812 ,1126 ;PA1828 ,1124 ;PA1844 ,1122 ;PA1860 ,1118 ;PA1876 ,1116 ;PA1892 ,1114 ;PA1908 ,1114 ;PA1924 ,1112 ;PA1940 ,1110 ;PA1956 ,1112 ;PA1972 ,1112 ;PA1988 ,1112 ;PA2004 ,1110 ;PA2020 ,1110 ;PA2036 ,1108 ;PA2052 ,1106 ;PA2068 ,1106 ;PA2084 ,1102 ;PA2100 ,1100 ;PA2116 ,1098 ;PA2132 ,1098 ;PA2148 ,1098 ;PA2164 ,1100 ;PA2180 ,1098 ;PA2196 ,1100 ;PA2212 ,1098 ;PA2228 ,1100 ;PA2244 ,1098 ;PA2260 ,1098 ;PA2276 ,1096 ;PA2292 ,1096 ;PA2308 ,1096 ;PA2324 ,1096 ;PA2340 ,1096 ;PA2356 ,1102 ;PA2372 ,1106 ;PA2388 ,1108 ;PA2404 ,1114 ;PA2420 ,1118 ;PA2436 ,1124 ;PA2452 ,1130 ;PA2468 ,1136 ;PA2484 ,1142 ;PA2500 ,1148 ;PA2516 ,1152 ;PA2532 ,1160 ;PA2548 ,1164 ;PA2564 ,1174 ;PA2580 ,1182 ;PA2596 ,1192 ;PA2612 ,1200 ;PA2628 ,1208 ;PA2644 ,1214 ;PA2660 ,1222 ;PA2676 ,1228 ;PA2692 ,1232 ;PA2708 ,1236 ;PA2724 ,1240 ;PA2740 ,1242 ;PA2756 ,1242 ;PA2772 ,1242 ;PA2788 ,1242 ;PA2804 ,1244 ;PA2820 ,1244 ;PA2836 ,1244 ;PA2852 ,1244 ;PA2868 ,1242 ;PA2884 ,1242 ;PA2900 ,1240 ;PA2916 ,1238 ;PA2932 ,1234 ;PA2948 ,1228 ;PA2964 ,1224 ;PA2980 ,1218 ;PA2996 ,1212 ;PA3012 ,1208 ;PA3028 ,1206 ;PA3044 ,1204 ;PA3060 ,1204 ;PA3076 ,1198 ;PA3092 ,1196 ;PA3108 ,1194 ;PA3124 ,1190 ;PA3140 ,1188 ;PA3156 ,1182 ;PA3172 ,1176 ;PA3188 ,1174 ;PA3204 ,1170 ;PA3220 ,1172 ;PA3236 ,1174 ;PA3252 ,1176 ;PA3268 ,1178 ;PA3284 ,1182 ;PA3300 ,1182 ;PA3316 ,1188 ;PA3332 ,1190 ;PA3348 ,1194 ;PA3364 ,1200 ;PA3380 ,1204 ;PA3396 ,1208 ;PA3412 ,1212 ;PA3428 ,1220 ;PA3444 ,1228 ;PU;PA0,0LT;;PU;PA0,0SP3;;PU;PA0,0LT;;PA3444 ,152 ;PD;PA244 ,152 ;PU;PA244 ,532 ;PD;PA3444 ,532 ;PU;PA3444 ,912 ;PD;PA244 ,912 ;PU;PA244 ,1292 ;PD;PA3444 ,1292 ;PU;PA3444 ,1672 ;PD;PA244 ,1672 ;PU;PA244 ,2052 ;PD;PA3444 ,2052 ;PU;PA3444 ,2432 ;PD;PA244 ,2432 ;PU;PA244 ,2812 ;PD;PA3444 ,2812 ;PU;PA3444 ,3192 ;PD;PA244 ,3192 ;PU;PA244 ,3572 ;PD;PA3444 ,3572 ;PU;PA3444 ,3952 ;PD;PA244 ,3952 ;PA244 ,152 ;PU;PA564 ,152 ;PD;PA564 ,3952 ;PU;PA884 ,3952 ;PD;PA884 ,152 ;PU;PA1204 ,152 ;PD;PA1204 ,3952 ;PU;PA1524 ,3952 ;PD;PA1524 ,152 ;PU;PA1844 ,152 ;PD;PA1844 ,3952 ;PU;PA2164 ,3952 ;PD;PA2164 ,152 ;PU;PA2484 ,152 ;PD;PA2484 ,3952 ;PU;PA2804 ,3952 ;PD;PA2804 ,152 ;PU;PA3124 ,152 ;PD;PA3124 ,3952 ;PU;PA3444 ,3952 ;PD;PA3444 ,152 LT;;PU;PA3444 ,2052 ;PD;PA244 ,2052 ;PA204 ,2072 ;PA204 ,2032 ;PA244 ,2052 SP1;;PU;PA0,0;PA0 ,4018 SR .981 , 1.519 ;;LBCH1 \'03;LT;;LBS\'03;;PA300 ,3986 ;LB11\'03;;PA420 ,4018 ;LB \'03;LT;;PA600 ,4018 ;LBlog MAG\'03;;PA1320 ,4018 ;LB10 dB/\'03;;PA1800 ,4018 ;LBREF 0 dB\'03;;PA0 ,24 ;LB    START    .300 000 MHz\'03;;PA2124 ,24 ;LBSTOP 3 000.000 000 MHz\'03;;PA1264 ,24 ;PA1800 ,24 LT;;PU;PA0,0;PA0 ,3584 SR .981 , 1.519 ;;LB \'03;;PA0 ,3456 ;LB  \'03;;PA0 ,1408 ;LB  \'03;;PA0 ,128 ;LB \'03;;PU;PA0,0;PA270 ,3824 SR .981 , 1.519 ;;LB\'03;;PA288 ,3824 ;PA288 ,3824 ;PA299 ,3890 ;PD;PA2 [... cut ...]
\par ++eoi 1
\par OUTPERRO
\par \cf2    31 ,"CAUTION: ADDRESSED TO TALK WITH NOTHING TO SAY"
\par \cf1 ++auto 0
\par ++loc
\par \cf3 Client 127.0.0.1 disconnected
\par Connected to client 127.0.0.1 , port = 8488
\par \cf1 ++eot_enable 0
\par ++ver
\par \cf2 Prologix version 6.101
\par \cf1 ++auto
\par \cf2 0
\par \cf1 ++auto 1
\par ++eos 0
\par ++mode 1
\par ++addr 16
\par ++eoi 1
\par OUTPPLOT;
\par \cf2\f1\'03;;;;DF;IM;\cf1 ++auto 0
\par \f0 ++loc
\par \cf3 Client 127.0.0.1 disconnected
\par 
\par }
 

Twilight-Logic, 2025-03-15:
Thank you for the updated information. It does seems to be literally reading just the first 11 characters. This will need further investigation.

Yrrah53, 2025-03-16:
Yes, that is exactly what I mean. Hope you can find a solution. Best regards,Harke(Send from my iPad4)

Twilight-Logic, 2025-03-26:
Apologies for the delay im getting back to you.

I set up a test rig to allow me to send the HP data as shown above over GPIB between to an AR488 adapter and the file was transmited whole. I was entering commands by hand, but also tested using a script. Of course, I am not testing with a HP8753 so the result may not be meaningful other than to show that it is possible to transmit the complete file over GPIB and have it received by the AR488 intact.

Is it possible for you to try with the version attached to this post please:

https://www.eevblog.com/forum/projects/ar488-arduino-based-gpib-adapter/msg5858619/#msg5858619

I would be nterested to know whether the same happens.

Thanks.

Yrrah53, 2025-03-28:
I loaded this version successfully: 0.52.29 24/03/2025. It runs normal with
my HP437B power meter, that I use to test the interface. Then I connect the
AR488 to my HP8753C vna. I run GPIB configurator and then 7470 Plotter
emulator. I still only get 11 bytes....
Best 73 de Harke

Twilight-Logic, 2025-03-30:
Was this without the last two commands, i.e. `++auto` and `++loc` ?

Yrrah53, 2025-03-31:
Its getting interesting. When I invoke a plot via Redirector it works. When
I try to get a plot via GPIB Configurator and 7470 emulator I only get the
11 bytes. When I try to connect the device via Redirector with the plotter
emulator I only get the usual 11 bytes. However,I can not repeat the first
successful result.
 These tests are shown in the txt file attached, with the exception of the
Toolkit test, of course.
If you need more data, just let me know, please.
Best regards,
Harke

{\rtf1\ansi\ansicpg1252\deff0\deflang1043{\fonttbl{\f0\fnil Tahoma;}{\f1\fnil\fcharset0 Tahoma;}}
{\colortbl ;\red0\green0\blue0;\red255\green0\blue0;\red0\green0\blue255;}
\viewkind4\uc1\pard\cf1\f0\fs20 Connecting...
\par Connected!
\par \cf2 ++ver
\par \cf3 AR488 GPIB controller, ver. 0.52.29, 24/03/2025
\par \cf2 ++id verstr Prologix AR488 HRK version 6.101
\par \cf3 Prologix AR488 HRK version 6.101
\par \cf2 ++savecfg
\par ++ver
\par \cf3 Prologix AR488 HRK version 6.101
\par \cf2 ++ver real
\par \cf3 AR488 GPIB controller, ver. 0.52.29, 24/03/2025
\par \cf2 IDN?
\par ++ver real
\par \cf3 AR488 GPIB controller, ver. 0.52.29, 24/03/2025
\par \cf2 ++ver
\par \cf3 Prologix AR488 HRK version 6.101
\par \cf2 ++addr 21
\par ++mode 1
\par ++auto 1
\par ++addr 16
\par IDN?
\par \cf3 HEWLETT PACKARD,8753C,0,4.13
\par \cf2 OUTPPLOT
\par \cf3\'03;;;;DF;IM;IP250,279,10250,7479;SC0 ,4095 ,0 ,4212 ;;PU;VS;SP1;;PU;PA0,0\f1 ;PA244 ,1980 ;PD;PA244 ,1980 ;PA244 ,1980 ;PA260 ,1976 ;PA276 ,1968 ;PA292 ,1964 ;PA308 ,1960 ;PA324 ,1958 ;PA340 ,1956 ;PA356 ,1956 ;PA372 ,1960 ;PA388 ,1964 ;PA404 ,1968 ;PA420 ,1970 ;PA436 ,1972 ;PA452 ,1972 ;PA468 ,1972 ;PA484 ,1972 ;PA500 ,1970 ;PA516 ,1970 ;PA532 ,1970 ;PA548 ,1974 ;PA564 ,1978 ;PA580 ,1984 ;PA596 ,1988 ;PA612 ,1986 ;PA628 ,1982 ;PA644 ,1976 ;PA660 ,1970 ;PA676 ,1972 ;PA692 ,1972 ;PA708 ,1976 ;PA724 ,1980 ;PA740 ,1980 ;PA756 ,1982 ;PA772 ,1982 ;PA788 ,1980 ;PA804 ,1978 ;PA820 ,1976 ;PA836 ,1972 ;PA852 ,1972 ;PA868 ,1972 ;PA884 ,1974 ;PA900 ,1978 ;PA916 ,1980 ;PA932 ,1978 ;PA948 ,1976 ;PA964 ,1974 ;PA980 ,1974 ;PA996 ,1972 ;PA1012 ,1970 ;PA1028 ,1966 ;PA1044 ,1962 ;PA1060 ,1960 ;PA1076 ,1960 ;PA1092 ,1964 ;PA1108 ,1970 ;PA1124 ,1972 ;PA1140 ,1972 ;PA1156 ,1970 ;PA1172 ,1966 ;PA1188 ,1962 ;PA1204 ,1958 ;PA1220 ,1956 ;PA1236 ,1954 ;PA1252 ,1952 ;PA1268 ,1952 ;PA1284 ,1954 ;PA1300 ,1956 ;PA1316 ,1956 ;PA1332 ,1954 ;PA1348 ,1948 ;PA1364 ,1942 ;PA1380 ,1938 ;PA1396 ,1936 ;PA1412 ,1936 ;PA1428 ,1936 ;PA1444 ,1936 ;PA1460 ,1936 ;PA1476 ,1934 ;PA1492 ,1934 ;PA1508 ,1932 ;PA1524 ,1936 ;PA1540 ,1936 ;PA1556 ,1938 ;PA1572 ,1938 ;PA1588 ,1938 ;PA1604 ,1936 ;PA1620 ,1934 ;PA1636 ,1934 ;PA1652 ,1932 ;PA1668 ,1932 ;PA1684 ,1932 ;PA1700 ,1934 ;PA1716 ,1940 ;PA1732 ,1944 ;PA1748 ,1948 ;PA1764 ,1948 ;PA1780 ,1948 ;PA1796 ,1946 ;PA1812 ,1946 ;PA1828 ,1948 ;PA1844 ,1950 ;PA1860 ,1952 ;PA1876 ,1954 ;PA1892 ,1956 ;PA1908 ,1954 ;PA1924 ,1956 ;PA1940 ,1958 ;PA1956 ,1960 ;PA1972 ,1962 ;PA1988 ,1962 ;PA2004 ,1962 ;PA2020 ,1962 ;PA2036 ,1962 ;PA2052 ,1962 ;PA2068 ,1962 ;PA2084 ,1964 ;PA2100 ,1964 ;PA2116 ,1962 ;PA2132 ,1964 ;PA2148 ,1966 ;PA2164 ,1968 ;PA2180 ,1966 ;PA2196 ,1962 ;PA2212 ,1960 ;PA2228 ,1958 ;PA2244 ,1960 ;PA2260 ,1964 ;PA2276 ,1968 ;PA2292 ,1968 ;PA2308 ,1968 ;PA2324 ,1962 ;PA2340 ,1958 ;PA2356 ,1954 ;PA2372 ,1952 ;PA2388 ,1954 ;PA2404 ,1958 ;PA2420 ,1958 ;PA2436 ,1960 ;PA2452 ,1962 ;PA2468 ,1964 ;PA2484 ,1962 ;PA2500 ,1956 ;PA2516 ,1950 ;PA2532 ,1942 ;PA2548 ,1936 ;PA2564 ,1934 ;PA2580 ,1934 ;PA2596 ,1932 ;PA2612 ,1932 ;PA2628 ,1930 ;PA2644 ,1930 ;PA2660 ,1932 ;PA2676 ,1934 ;PA2692 ,1934 ;PA2708 ,1932 ;PA2724 ,1926 ;PA2740 ,1920 ;PA2756 ,1914 ;PA2772 ,1912 ;PA2788 ,1914 ;PA2804 ,1918 ;PA2820 ,1924 ;PA2836 ,1926 ;PA2852 ,1928 ;PA2868 ,1928 ;PA2884 ,1926 ;PA2900 ,1924 ;PA2916 ,1920 ;PA2932 ,1914 ;PA2948 ,1912 ;PA2964 ,1916 ;PA2980 ,1920 ;PA2996 ,1924 ;PA3012 ,1928 ;PA3028 ,1932 ;PA3044 ,1934 ;PA3060 ,1934 ;PA3076 ,1936 ;PA3092 ,1936 ;PA3108 ,1938 ;PA3124 ,1940 ;PA3140 ,1940 ;PA3156 ,1940 ;PA3172 ,1940 ;PA3188 ,1942 ;PA3204 ,1942 ;PA3220 ,1942 ;PA3236 ,1940 ;PA3252 ,1940 ;PA3268 ,1940 ;PA3284 ,1942 ;PA3300 ,1944 ;PA3316 ,1944 ;PA3332 ,1944 ;PA3348 ,1946 ;PA3364 ,1948 ;PA3380 ,1952 ;PA3396 ,1956 ;PA3412 ,1960 ;PA3428 ,1960 ;PA3444 ,1960 ;PU;PA0,0LT;;PU;PA0,0SP3;;PU;PA0,0LT;;PA3444 ,152 ;PD;PA244 ,152 ;PU;PA244 ,532 ;PD;PA3444 ,532 ;PU;PA3444 ,912 ;PD;PA244 ,912 ;PU;PA244 ,1292 ;PD;PA3444 ,1292 ;PU;PA3444 ,1672 ;PD;PA244 ,1672 ;PU;PA244 ,2052 ;PD;PA3444 ,2052 ;PU;PA3444 ,2432 ;PD;PA244 ,2432 ;PU;PA244 ,2812 ;PD;PA3444 ,2812 ;PU;PA3444 ,3192 ;PD;PA244 ,3192 ;PU;PA244 ,3572 ;PD;PA3444 ,3572 ;PU;PA3444 ,3952 ;PD;PA244 ,3952 ;PA244 ,152 ;PU;PA564 ,152 ;PD;PA564 ,3952 ;PU;PA884 ,3952 ;PD;PA884 ,152 ;PU;PA1204 ,152 ;PD;PA1204 ,3952 ;PU;PA1524 ,3952 ;PD;PA1524 ,152 ;PU;PA1844 ,152 ;PD;PA1844 ,3952 ;PU;PA2164 ,3952 ;PD;PA2164 ,152 ;PU;PA2484 ,152 ;PD;PA2484 ,3952 ;PU;PA2804 ,3952 ;PD;PA2804 ,152 ;PU;PA3124 ,152 ;PD;PA3124 ,3952 ;PU;PA3444 ,3952 ;PD;PA3444 ,152 LT;;PU;PA3444 ,2052 ;PD;PA244 ,2052 ;PA204 ,2072 ;PA204 ,2032 ;PA244 ,2052 SP1;;PU;PA0,0;PA0 ,4018 SR .981 , 1.519 ;;LBCH1 \'03;LT;;LBS\'03;;PA300 ,3986 ;LB11\'03;;PA420 ,4018 ;LB \'03;LT;;PA600 ,4018 ;LBlog MAG\'03;;PA1320 ,4018 ;LB10 dB/\'03;;PA1800 ,4018 ;LBREF 0 dB\'03;;PA0 ,24 ;LB    START    .300 000 MHz\'03;;PA2124 ,24 ;LBSTOP 3 000.000 000 MHz\'03;;PA1264 ,24 ;PA1800 ,24 LT;;PU;PA0,0;PA0 ,3584 SR .981 , 1.519 ;;LB \'03;;PA0 ,3456 ;LB  \'03;;PA0 ,1408 ;LB  \'03;;PA0 ,128 ;LB \'03;;PU;PA0,0;PA270 ,3824 SR .981 , 1.519 ;;LB\'03;;PA288 ,3824 ;PA288  [... cut ...]
\par \cf3\'03;;;;DF;IM;\cf2 ++mode 1
\par ++mode 1++auto 1
\par ++mode 1
\par ++auto 1
\par ++loc
\par ++addr 16
\par ++ver
\par \cf3 Prologix AR488 HRK version 6.101
\par \cf2 IDN?
\par \cf3 HEWLETT PACKARD,8753C,0,4.13
\par \cf2 OUTPPLOT
\par \cf3\'03;;\f1 ;;DF;IM;\cf2 ++loc
\par \f0 ++auto
\par \cf3 1
\par \cf2 IDN?
\par \cf3 HEWLETT PACKARD,8753C,0,4.13
\par \cf2 OUTPLOT
\par OUTPLOT
\par ++auto 1
\par IDN?
\par \cf3 HEWLETT PACKARD,8753C,0,4.13
\par \cf2 OUTPPLOT
\par \cf3\f1\'03;;;;DF;IM;\cf0\f0 
\par }
 

Yrrah53, 2025-03-31:
Another observation: I just started from scratch: switch AR488 and HP8753C
off and on again. From Redirector:
++address 16
++auto 1
++mode 1
then: IDN? give the correct response.
OUTPPLOT: gives the complete plot output.
OUTPPLOT again: same result....
Starting from scratch seems required.
Best regards,
Harke

Twilight-Logic, 2025-04-01:
Try setting `++read_tmo_ms 10000` at the start:

```
++ver
++read_tmo_ms 10000
++mode 1
++auto 1
++addr 16
*IDN?
OUTPLOT
```

Yrrah53, 2025-04-02:
Thanks, I tested this sequence with Redirector: works fine. I saved the
configuration. Then a cold start: works fine. Then I started Redirector
with the 7470 emulator: same trouble.
Please refer to annex.
Please let me know if you want further tests. It seems that as soon as 7470
emulator gets involved it starts getting weird.
Best regards,
Harke

{\rtf1\ansi\ansicpg1252\deff0\deflang1043{\fonttbl{\f0\fnil Tahoma;}{\f1\fnil\fcharset0 Tahoma;}}
{\colortbl ;\red0\green0\blue0;\red255\green0\blue0;\red0\green0\blue255;}
\viewkind4\uc1\pard\cf1\f0\fs20 Connecting...
\par Connected!
\par \cf2 ++ver
\par \cf3 Prologix AR488 HRK version 6.101
\par \cf2 ++ver real
\par \cf3 AR488 GPIB controller, ver. 0.52.29, 24/03/2025
\par \cf2 IDN?
\par \cf3 HEWLETT PACKARD,8753C,0,4.13
\par \cf2 OUTPPLOT
\par \cf3\'03;;;;DF;IM;IP250,279,10250,7479;SC0 ,4095 ,0 ,4212 ;;PU;VS;SP1;\f1 ;PU;PA0,0;PA244 ,1978 ;PD;PA244 ,1978 ;PA244 ,1978 ;PA260 ,1976 ;PA276 ,1968 ;PA292 ,1966 ;PA308 ,1960 ;PA324 ,1958 ;PA340 ,1956 ;PA356 ,1958 ;PA372 ,1960 ;PA388 ,1964 ;PA404 ,1968 ;PA420 ,1972 ;PA436 ,1972 ;PA452 ,1974 ;PA468 ,1972 ;PA484 ,1972 ;PA500 ,1970 ;PA516 ,1970 ;PA532 ,1970 ;PA548 ,1974 ;PA564 ,1978 ;PA580 ,1984 ;PA596 ,1986 ;PA612 ,1986 ;PA628 ,1982 ;PA644 ,1976 ;PA660 ,1972 ;PA676 ,1972 ;PA692 ,1972 ;PA708 ,1976 ;PA724 ,1980 ;PA740 ,1980 ;PA756 ,1982 ;PA772 ,1982 ;PA788 ,1980 ;PA804 ,1978 ;PA820 ,1976 ;PA836 ,1972 ;PA852 ,1970 ;PA868 ,1972 ;PA884 ,1976 ;PA900 ,1978 ;PA916 ,1978 ;PA932 ,1978 ;PA948 ,1976 ;PA964 ,1974 ;PA980 ,1974 ;PA996 ,1972 ;PA1012 ,1970 ;PA1028 ,1968 ;PA1044 ,1962 ;PA1060 ,1960 ;PA1076 ,1960 ;PA1092 ,1964 ;PA1108 ,1970 ;PA1124 ,1972 ;PA1140 ,1972 ;PA1156 ,1970 ;PA1172 ,1966 ;PA1188 ,1962 ;PA1204 ,1960 ;PA1220 ,1956 ;PA1236 ,1954 ;PA1252 ,1952 ;PA1268 ,1952 ;PA1284 ,1954 ;PA1300 ,1956 ;PA1316 ,1958 ;PA1332 ,1954 ;PA1348 ,1950 ;PA1364 ,1942 ;PA1380 ,1938 ;PA1396 ,1936 ;PA1412 ,1936 ;PA1428 ,1936 ;PA1444 ,1936 ;PA1460 ,1936 ;PA1476 ,1934 ;PA1492 ,1934 ;PA1508 ,1934 ;PA1524 ,1936 ;PA1540 ,1938 ;PA1556 ,1938 ;PA1572 ,1940 ;PA1588 ,1938 ;PA1604 ,1938 ;PA1620 ,1934 ;PA1636 ,1934 ;PA1652 ,1932 ;PA1668 ,1932 ;PA1684 ,1932 ;PA1700 ,1934 ;PA1716 ,1940 ;PA1732 ,1944 ;PA1748 ,1948 ;PA1764 ,1948 ;PA1780 ,1948 ;PA1796 ,1946 ;PA1812 ,1946 ;PA1828 ,1948 ;PA1844 ,1950 ;PA1860 ,1954 ;PA1876 ,1954 ;PA1892 ,1956 ;PA1908 ,1956 ;PA1924 ,1956 ;PA1940 ,1958 ;PA1956 ,1960 ;PA1972 ,1962 ;PA1988 ,1964 ;PA2004 ,1964 ;PA2020 ,1962 ;PA2036 ,1962 ;PA2052 ,1962 ;PA2068 ,1964 ;PA2084 ,1964 ;PA2100 ,1964 ;PA2116 ,1964 ;PA2132 ,1964 ;PA2148 ,1966 ;PA2164 ,1968 ;PA2180 ,1966 ;PA2196 ,1962 ;PA2212 ,1960 ;PA2228 ,1960 ;PA2244 ,1960 ;PA2260 ,1964 ;PA2276 ,1968 ;PA2292 ,1970 ;PA2308 ,1968 ;PA2324 ,1962 ;PA2340 ,1958 ;PA2356 ,1954 ;PA2372 ,1952 ;PA2388 ,1954 ;PA2404 ,1958 ;PA2420 ,1960 ;PA2436 ,1962 ;PA2452 ,1964 ;PA2468 ,1964 ;PA2484 ,1962 ;PA2500 ,1958 ;PA2516 ,1950 ;PA2532 ,1942 ;PA2548 ,1938 ;PA2564 ,1936 ;PA2580 ,1934 ;PA2596 ,1934 ;PA2612 ,1934 ;PA2628 ,1932 ;PA2644 ,1932 ;PA2660 ,1932 ;PA2676 ,1936 ;PA2692 ,1936 ;PA2708 ,1932 ;PA2724 ,1926 ;PA2740 ,1920 ;PA2756 ,1914 ;PA2772 ,1912 ;PA2788 ,1914 ;PA2804 ,1918 ;PA2820 ,1924 ;PA2836 ,1928 ;PA2852 ,1928 ;PA2868 ,1928 ;PA2884 ,1926 ;PA2900 ,1924 ;PA2916 ,1920 ;PA2932 ,1916 ;PA2948 ,1914 ;PA2964 ,1916 ;PA2980 ,1920 ;PA2996 ,1926 ;PA3012 ,1930 ;PA3028 ,1932 ;PA3044 ,1934 ;PA3060 ,1936 ;PA3076 ,1936 ;PA3092 ,1936 ;PA3108 ,1938 ;PA3124 ,1942 ;PA3140 ,1942 ;PA3156 ,1942 ;PA3172 ,1942 ;PA3188 ,1942 ;PA3204 ,1942 ;PA3220 ,1942 ;PA3236 ,1942 ;PA3252 ,1940 ;PA3268 ,1940 ;PA3284 ,1942 ;PA3300 ,1944 ;PA3316 ,1946 ;PA3332 ,1944 ;PA3348 ,1946 ;PA3364 ,1950 ;PA3380 ,1954 ;PA3396 ,1958 ;PA3412 ,1960 ;PA3428 ,1962 ;PA3444 ,1962 ;PU;PA0,0LT;;PU;PA0,0SP3;;PU;PA0,0LT;;PA3444 ,152 ;PD;PA244 ,152 ;PU;PA244 ,532 ;PD;PA3444 ,532 ;PU;PA3444 ,912 ;PD;PA244 ,912 ;PU;PA244 ,1292 ;PD;PA3444 ,1292 ;PU;PA3444 ,1672 ;PD;PA244 ,1672 ;PU;PA244 ,2052 ;PD;PA3444 ,2052 ;PU;PA3444 ,2432 ;PD;PA244 ,2432 ;PU;PA244 ,2812 ;PD;PA3444 ,2812 ;PU;PA3444 ,3192 ;PD;PA244 ,3192 ;PU;PA244 ,3572 ;PD;PA3444 ,3572 ;PU;PA3444 ,3952 ;PD;PA244 ,3952 ;PA244 ,152 ;PU;PA564 ,152 ;PD;PA564 ,3952 ;PU;PA884 ,3952 ;PD;PA884 ,152 ;PU;PA1204 ,152 ;PD;PA1204 ,3952 ;PU;PA1524 ,3952 ;PD;PA1524 ,152 ;PU;PA1844 ,152 ;PD;PA1844 ,3952 ;PU;PA2164 ,3952 ;PD;PA2164 ,152 ;PU;PA2484 ,152 ;PD;PA2484 ,3952 ;PU;PA2804 ,3952 ;PD;PA2804 ,152 ;PU;PA3124 ,152 ;PD;PA3124 ,3952 ;PU;PA3444 ,3952 ;PD;PA3444 ,152 LT;;PU;PA3444 ,2052 ;PD;PA244 ,2052 ;PA204 ,2072 ;PA204 ,2032 ;PA244 ,2052 SP1;;PU;PA0,0;PA0 ,4018 SR .981 , 1.519 ;;LBCH1 \'03;LT;;LBS\'03;;PA300 ,3986 ;LB11\'03;;PA420 ,4018 ;LB \'03;LT;;PA600 ,4018 ;LBlog MAG\'03;;PA1320 ,4018 ;LB10 dB/\'03;;PA1800 ,4018 ;LBREF 0 dB\'03;;PA0 ,24 ;LB    START    .300 000 MHz\'03;;PA2124 ,24 ;LBSTOP 3 000.000 000 MHz\'03;;PA1264 ,24 ;PA1800 ,24 LT;;PU;PA0,0;PA0 ,3584 SR .981 , 1.519 ;;LB \'03;;PA0 ,3456 ;LB  \'03;;PA0 ,1408 ;LB  \'03;;PA0 ,128 ;LB \'03;;PU;PA0,0;PA270 ,3824 SR .981 , 1.519 ;;LB\'03;;PA288 ,3824 ;PA288 ,3824 ;PA299 ,3890 ;PD;PA288 ,3890 ;PA288 ,3824 ;PA316 ,3824 ;PU;PA299 ,3842 ;PD;PA311 ,3890 ;PU;PA305 ,3866 ;PD;PA322 ,3866 ;PA316 ,3842 ;PU;PA348 ,3824 ;PA330 ,3824 ;PA330 ,3824 ;PD;PA341 ,3866 ;PA353 ,3866 ;PA358 ,3860 ;PA353 ,3842 ;PA335 ,3842 ;PU;PA330 ,3890 ;PD;PA370 ,3890 ;PA370 ,3824 ;PA341 ,3824 ;PU;PA390 ,3824 ;PA0 ,2688 ;LB \'03;;PA0 ,3328 ;LB \'03;;PA0 ,3072 ;LB \'03;;PA0 ,2944 ;LB \'03;;PA0 ,2816 ;LB \'03;;PA0 ,3200 ;LB \'03;;PA0 ,2560 ;LB \'03;;PA0 ,2432 ;LB \'03;;PA0 ,2304 ;LB \'03;;PU;PA0,0LT;;PA270 ,3444 SR .98 [... cut ...]
\par \cf2 ++auto 0
\par ++loc
\par \cf1 Client 127.0.0.1 disconnected
\par Connected to client 127.0.0.1 , port = 8488
\par \cf2 ++eot_enable 0
\par ++ver
\par \cf3 Prologix AR488 HRK version 6.101
\par \cf2 ++auto
\par \cf3 0
\par \cf2 ++auto 1
\par ++eos 0
\par ++mode 1
\par ++addr 16
\par ++eoi 1
\par OUTPPLOT;
\par \cf3\f1\'03;;;;DF;IM;\cf2 ++auto 0
\par \f0 ++loc
\par \cf1 Client 127.0.0.1 disconnected
\par ComPort A disconnected
\par connect.ini restored to previous state.
\par \cf0 
\par }
 

Twilight-Logic, 2025-04-03:
That is interesting. So it does look like a timeout problem. We increased it from 1200ms (1.2s) to 10000ms (10s). Possibly it didn't need increasing that much, but it seems that worked for Redirector. You might be able to reduce this to 5000 or 3000 but you would have to experiment.

Looking at 7470.ini, I notice that it too has similar settings and I am wondering whether one of them is sending an ++read+tmo_ms command and overriding the AR488 setting. These are the two settings I am referring to:

```
async_GPIB_timeout_ms 1000
serial_read_dropout   500
```

I am thinking it might be worth increasing these as an experiment. Maybe first try increasing `async_GPIB_timeout_ms`  from 1000 to 5000 to see what happens. If that's doesn't change anything, then try increasing `async_GPIB_timeout_ms` as well. I would be interested to know whether this makes any difference.

Can you tell me where I can get Redirector from? I couldn't find it online?

Yrrah53, 2025-04-03:
Thanks for your reply. As I will be travelling tomorrow for more than a
week I do not have the time for further experiments. Will come back to you
later.
You can certainly obtain Redirector from Tom Iphi, Tom Baier, you know from
EEVBlog.
Best regards,
Harke

Yrrah53, 2025-04-29:
Ok, so I tried with increased timing parameters as requested (in 7470.ini).
No luck: still only 11 bytes received by 7470.exe. In the meantime I did
some extensive testing with Tom and Redirector and we could not find the
problem. Via BT or USB does not make a difference. I made a small Python
script to read IDN? and a plot (OUTPPLOT) from the HP8753C. Basically it
works but the read result is messed up: the ++read output seems to be
delayed by one ++read statement. So it happens that the plot perfectly
shows up in the next run.... (++auto is set to 0)
Magnus/Kofen will also perform a test with a HP8753C shortly....

Yrrah53, 2025-04-29:
Another observation. I have two versions (in fact three, also with a LAN
interface). The version of 18/03/2024 has a BT interface and seems to work
better than the latest version with USB??? I can get the full plot into the
Log Traffic screen of Redirector.
I'm afraid I am running out of ideas....
Best regards,
Harke

Twilight-Logic, 2025-05-01:
Harke, thanks for the feedback. That seems to tie in with Toms comment on EEVBLOG about a friend who has a problem with auto-read and obtaining a plot and the same USB vs Bluetooth problem.... Tom did post a comment later saying he had got it working both ways but I am not sure at this point whether it was a fluke or whether he has it working reliably now.

> Basically it works but the read result is messed up: the ++read output seems to be
> delayed by one ++read statement. So it happens that the plot perfectly
> shows up in the next run.... 

That sounds like the plot is not yet ready the first time you run `++read` and since no data is available yet, it returns nothing. By the time you run `++read` a second time, its ready and the data is sent. What happens if you add a delay in your script (say 10-15 seconds) before sending the '++read' command?

Magnus has also been in touch regarding a problem with the Ethernet version. Hopefully I will get an Ethernet board to test soon. He also mentioned some mod to emulate a plotter which might be useful as well. I need to find some way to emulate these conditions. In the meantime, let me know whether adding the delay to the script makes any difference, or whether you have tried that already?

Yrrah53, 2025-05-01:
At this point Tom and I are rather exhausted. We did many tests. To give you an example: the attached Python script works perfect via LAN and messes things up via USB. As you can see I added a delay after the OUTPPLOT statement. 3 or 5 sec: no difference. 5 sec should be enough. In Redirector it goes wrong as soon as 7470.exe is involved: only the first 11 or 12 bytes are received. Regardless the interface. Hope you have a clue?

`# try out some gpib data collection using the Prologix AR488 gpib-lan controller
# Harke Smits
# Test for HP8753C vna @ GPIB address 16
# serial data format: bytes....----> str.encode("..")
# Sept 2024/October 2024: SRQ debugged....
# Jan 2025: Linux Mint/Windows version
# April 2025: LAN version and some improvements (tnx to Magnus!)
# werkt met AR488, maar met enorme delay
# Works for LAN: AR488 08/04/2025 (via Magnus). The USB version is: 24/03/2025 (Github)

import time
import socket
import serial
from sys import platform

ethernet = False  # Set True for Ethernet, False for serial
ip_address = "192.168.178.74"  # Set your Prologix IP here
tcp_port = 1234

gpib_address = 16  # HP8753C vna default GPIB address

# Serial or socket interface
ser = None

def connect():
    global ser
    if ethernet:
        print(f"Using Ethernet connection to {ip_address}:{tcp_port}")
        ser = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        ser.settimeout(8)
        ser.connect((ip_address, tcp_port))
    else:
        if platform == "linux":
            interface_id = '/dev/ttyUSB0'
            print("Platform is Linux")
        elif platform == "win32":
            interface_id = 'COM6'
            print("Platform is Windows")
        else:
            raise RuntimeError("Unsupported platform")
        ser = serial.Serial(interface_id, baudrate=115200, rtscts=0, timeout=1)

def ser_write(data: bytes):
    if ethernet:
        ser.sendall(data + b'\n') # Adding \n so that the adapter knows it is end of command(True prologix behaviour)
    else:
        ser.write(data)

def ser_readline() -> str:
    if ethernet:
        line = b""
        while not line.endswith(b"\n"): # r was n
            chunk = ser.recv(1)
            if not chunk:
                break
            line += chunk
        return line.decode().strip()
    else:
        return ser.readline().decode().strip()

def gpib_init() -> None:
    ser_write(b"++mode 1\r")
    time.sleep(0.1)
    ser_write(b"++ifc\r")
    time.sleep(0.1)
    ser_write(b"++auto 0\r")
    time.sleep(0.1)
    ser_write(b"++eoi 0\r")
    time.sleep(0.1)
    ser_write(b"++read_tmo_ms 500")
    time.sleep(0.1)
    ser_write(b"++ver\r")
    print(f"Prologix version: {ser_readline()}")

def gpib_read(addr=gpib_address):
    ser_write(f"++addr {addr}\r".encode())
    time.sleep(0.1)
    ser_write(b"++read ") #eoi\n")
    return ser_readline()

def gpib_write(gpibstr, addr=gpib_address):
    ser_write(f"++addr {addr}\r".encode())
    time.sleep(0.1)
    ser_write(f"{gpibstr}\r".encode())

def check_srq() -> bool:
    ser_write(b"++srq\r")
    result = ser_readline()
    return result == "1"

# ---- MAIN ----
connect()
gpib_init()

gpib_write("*RST;IDN?")
time.sleep(1)
# gpib_write("IDN?")
print(f"ID = {gpib_read()}")
gpib_write("CLS")
time.sleep(0.1)
gpib_write("STB?")
value = gpib_read()
print(f"\nStatus byte: {value}")
gpib_write("OUTPPLOT")
time.sleep(3)
display=gpib_read()
# time.sleep(10)
print(f"Plot 1: {display}")
# display=gpib_read()
# time.sleep(10)
# print(f"Plot 2: {display}")

print("\nEnd.")`

Twilight-Logic, 2025-05-01:
Thank you both for your efforts. It sounds like you have done a lot of work on this!

Thank you for including the Python script. It seems the Github Markdown parser has scrambled it a bit. I think I have reconstructed it but not entirely sure. Could you attach is as a file please?

The LAN interface arrived today so I can now test and compare both LAN and USB. I would like to see whether I can replicate what you are getting with the script.  The only way to do this short of actually buying or somehow getting access to a HP8753C is to emulate sending of the plot. I can get the command sequences from the script and from your comment it sounds like the delay to compose and send the plot is no more than 3 or 4  seconds. I am going to need a bit of time to set this up as I need to make changes to the device node emulator. I did try this previously, but without any delays.

I probably need to ask Tom for some help with setting up Redirector and connecting it up to the KE5FX 7470 plotter program so that I set it up exactly as you have done.

I note that the Python program also runs on Linux, so I presume that you conducted tests on both Windows and Linux?

BTW, I did see an 8757C for sale for a more reasonable sum but even that was not cheap and I don't know whether it would give me the same result. Maybe I could buy and and sell it later. Will have to think about that.

Yrrah53, 2025-05-02:
Yes Tom and I did a lot of testing and I am grateful for Tom trying to help
me out. In the end our ideas were exhausted....
The Python file attached is a derivative of a script I made to control the
HP437B RF Power Meter. The --Main-- part has been changed to test with a
HP8753C vna. It's very basic. I can see from the print if the plot is ok,
it is not even saved. Though that is fairly easy, of course. The original
script has been tested with three adapters and W10/Linux and works.
It seems there are two major issues:
the USB version messes with the buffers. The expected ++read data output
comes one or more read statements later. Sometimes the plot shows as the
first output of the next run. I have not seen this with the lan version
(latest sw version from Magnus). Same Python script of course.
Then there is a general issue related to 7470.exe. If I simulate the
command sequence manually, or in Redirector Bulk mode: no problem, I get
the full plot as data in the monitor. As soon as it tries to connect with
7470.exe only the first few bytes are shown. Either via Redirector or via
GPIB configurator.
I hope you can get to a solution. If you need more data or more testing,
please let me know.

The HP8757C is a different beast, and though interesting as I have a 8756A,
not relevant for simulating a HP8753C vna.

I'll be travelling the next week, but available via mail.
Best regards,
Harke

Yrrah53, 2025-07-30:
Actually, I solved this for the HP8753C vna by a multiple of read statements, in fact 5. Somehow the plot data is released in chunks? It's ugly but it works. I have a Python script that fetches the plot data and then calls 7470.exe to show the plot. Works well.