Twilight-Logic/AR488 #51: Connecting to NI MAX

YYYYYJS, 2025-04-08:
Hi, may I ask—if I use NI's GPIB-USB-HS to connect to the AR488, will NI MAX be able to recognize it successfully?

Twilight-Logic, 2025-04-08:
I don't have a definite answer for this as I have not been able to test it with a GPIB-USB-HS adapter. The AR488 doesn't yet recognise NI's high speed protocol released in the updated standard 488.1 but the NI adapter should be able to drop back to standard GPIB and be able to interoperate. I would be interested in knowing the answer to this myself.

YYYYYJS, 2025-04-09:
谢谢，我现在正在尝试。如果有结果我再告诉你

> 我对此没有明确的答案，因为我无法用 GPIB-USB-HS 适配器对其进行测试。AR488 尚不认可更新后的标准 488.1 中发布的 NI 高速协议，但 NI 适配器应该能够退回到标准 GPIB 并能够互作。我很想知道自己这个问题的答案。

YYYYYJS, 2025-04-09:
Thank you, I'm trying it now. I'll let you know if I get any results.

> I don't have a definite answer for this as I have not been able to test it with a GPIB-USB-HS adapter. The AR488 doesn't yet recognise NI's high speed protocol released in the updated standard 488.1 [...]

TekCastPork, 2025-09-26:
I'm trying the same thing, but also with HS+ adapters, and I've been finding that with a configuration which should set the ar488 as a device with PAD 1 AND NO_SAD (changing `cmode` to 1, ontop of `MACRO_0` executing a `++mode 0` just in case) NI-MAX reports finding too many devices on the bus when i scan for instruments, despite the only thing being connected is my ar488.

I can see my ar488 receiving data from the bus when it scans via the debug console. This is on an atmega328pb on a custom board, but it does use the pinout config of the standard UNO layout, utilizing SN7516x buffers.

YYYYYJS, 2025-09-27:
Thank you for your detailed email describing the issue you encountered. It was very helpful to read about your specific setup and troubleshooting steps.
I'm quite interested in the "device overload" problem you faced with NI-MAX during the instrument scan. I was wondering if you have had any success in resolving it since your last message? I would be very grateful to hear about any new findings or potential solutions you might have discovered.
Please let me know if there is any additional information from my end that could help in diagnosing the problem.
Best regards,
Jiasheng Yin
---- Replied Message ----
From ***@***.***> Date 09/26/2025 23:28 To ***@***.***> Cc ***@***.***>***@***.***> Subject Re: [Twilight-Logic/AR488] Connecting to NI MAX (Issue #51) 
TekCastPork left a comment (Twilight-Logic/AR488#51)
I'm trying the same thing, but also with HS+ adapters, and I've been finding that with a configuration which should set the ar488 as a device with PAD 1 AND NO_SAD (changing cmode to 1, ontop of MACRO_0 executing a ++mode 0 just in case) NI-MAX reports finding too many devices on the bus when i scan for instruments, despite the only thing being connected is my ar488.
I can see my ar488 receiving data from the bus when it scans via the debug console. This is on an atmega328pb on a custom board, but it does use the pinout config of the standard UNO layout, utilizing SN7516x buffers.

TekCastPork, 2025-09-27:
I havent yet, its also for a work thing I cant really go too deep into what i need it for and also wont be able to touch it again until after the weekend.

From what i can gather though it looks like it may be pulling NDAC low each time my hs+ pulls ATN low.

This makes sense for the 488.2 find listeners function behavior, but it seems to be doing it for every listener address test, which likely explains why it was found 31 times (which is too many, out of 32 possible primary addresses addr 0 is the controller and 31 is a special address invalid for instruments, so it can only use between addr 1 and addr 30).

Im going to sniff the data lines more in depth when I work on it again to hopefully confirm this theory.

YYYYYJS, 2025-09-27:
Thank you for the update and sharing your detailed analysis. Your reasoning about the NDAC/ATN interaction and address range makes a lot of sense and is very helpful.
I'll also be sure to share any findings from my end as I continue my own testing. Looking forward to your further findings next week. Enjoy your weekend.
---- Replied Message ----
From ***@***.***> Date 09/27/2025 11:19 To ***@***.***> Cc ***@***.***>***@***.***> Subject Re: [Twilight-Logic/AR488] Connecting to NI MAX (Issue #51) 
TekCastPork left a comment (Twilight-Logic/AR488#51)
I havent yet, its also for a work thing I cant really go too deep into what i need it for and also wont be able to touch it again until after the weekend.
From what i can gather though it looks like it may be pulling NDAC low each time my hs+ pulls ATN low.
This makes sense for the 488.2 find listeners function behavior, but it seems to be doing it for every listener address test, which likely explains why it was found 31 times (which is too many, out of 32 possible primary addresses addr 0 is the controller and 31 is a special address invalid for instruments, so it can only use between addr 1 and addr 30).
Im going to sniff the data lines more in depth when I work on it again to hopefully confirm this theory.