Twilight-Logic/AR488 #56: GPIB Hangup Issues

kmuthugtk, 2025-04-23:
During a test run, I suddenly encountered a GPIB hang-up issue on the tester side, which resulted in missing data. Please refer to the attached screenshot for details. I would appreciate it if anyone could help provide a solution.

[image]

Twilight-Logic, 2025-04-25:
I will need some further information.

What hardware were you using?

What program is this screeenshot from?

Twilight-Logic, 2025-07-13:
Closing due the lack of further information. If the issue still persists in version 0.53.20, then please feel free to re-open the case and provide the requested information.

Thank you.

closed 2025-07-13 by Twilight-Logic