Twilight-Logic/AR488 #58: First Character is missing on *IDN?

WilheJo, 2025-04-29:
On my SMY01, the first character is missing since 0.52.26.

See log below.

BTW: DEBUG_GPIBbus_RECEIVE fails to compile as well!

Log 0.52.26:

AR488 GPIB controller, ver. 0.52.26, 16/03/2025
Verbose: ON
> 
> ++addr 28

PRI address set to: 28
SEC address set to: unset
> 
> *IDN?
AR488_GPIBbus.cpp:861 (setControls) > Set GPIB lines for sending a command
AR488_GPIBbus.cpp:886 (setControls) > Set GPIB lines for writing data
AR488_GPIBbus.cpp:707 (sendData) > write data mode is set.
AR488_GPIBbus.cpp:708 (sendData) > Begin send loop ->
*IDN?AR488_GPIBbus.cpp:738 (sendData) > <- End of send loop.
AR488_GPIBbus.cpp:762 (sendData) > appended CRLF
AR488_GPIBbus.cpp:849 (setControls) > Set GPIB lines to idle state
AR488_GPIBbus.cpp:776 (sendData) > done.
AR488_GPIBbus.cpp:861 (setControls) > Set GPIB lines for sending a command
> 
> ++read

AR488_GPIBbus.cpp:874 (setControls) > Set GPIB lines for reading data
ROHDE&SCHWARZ,SMY01,839.069/002,2.02
AR488_GPIBbus.cpp:861 (setControls) > Set GPIB lines for sending a command
AR488_GPIBbus.cpp:849 (setControls) > Set GPIB lines to idle state
> 
> 

updated to latest upstream (including 2ee1956d9bd6b1539d7557ddb550db8c9a797484)

AR488 GPIB controller, ver. 0.53.04 (JW), 13/04/2025
Verbose: ON
> 
> ++addr 28

PRI address set to: 28
SEC address set to: unset
> 
> *IDN?
AR488_GPIBbus.cpp:862 (setControls) > Set GPIB lines for sending a command
AR488_GPIBbus.cpp:887 (setControls) > Set GPIB lines for writing data
AR488_GPIBbus.cpp:707 (sendData) > write data mode is set.
AR488_GPIBbus.cpp:708 (sendData) > Begin send loop ->
*IDN?AR488_GPIBbus.cpp:739 (sendData) > <- End of send loop.
AR488_GPIBbus.cpp:763 (sendData) > appended CRLF
AR488_GPIBbus.cpp:850 (setControls) > Set GPIB lines to idle state
AR488_GPIBbus.cpp:777 (sendData) > done.
AR488_GPIBbus.cpp:862 (setControls) > Set GPIB lines for sending a command
> 
> ++read

AR488_GPIBbus.cpp:875 (setControls) > Set GPIB lines for reading data
OHDE&SCHWARZ,SMY01,839.069/002,2.02
AR488_GPIBbus.cpp:850 (setControls) > Set GPIB lines to idle state
AR488_GPIBbus.cpp:862 (setControls) > Set GPIB lines for sending a command
> 
>

Twilight-Logic, 2025-04-29:
Thanks for the report. I will compare the code bases to see whether I can spot anything. I take it that this is happening without debugs enabled?

I will also check why DEBUG_GPIBbus_RECEIVE fails to compile.

WilheJo, 2025-04-29:
Yes. Happens both, with and without debug options.

Twilight-Logic, 2025-04-29:
I have tidied up the debug compile problem and checked the other debug options for similar issues.

Of course the more important problem is the dropping of the character.. Could you tell me which board you are using and which instrument please?

WilheJo, 2025-04-29:
Upps... Missed this little piece of information...

It's running on a esp32s2 with 7516x buffers.

Code: https://github.com/WilheJo-Org1/AR488

Schematic: https://www.diyemc.com/Products/Esp32s2Ar488/Rev4-Schematic.pdf

The code in the repo includes besides the layout some changes to have cdc with tiny-usb instead of the cdc-on-boot serial to have a serial number based on the factory programmed MAC addresses.
I checked without these as well... Same behaviour.

Twilight-Logic, 2025-04-29:
I wasn't expecting an ESP32 so that is interesting. I see that you are using the `ESP32_Wilhelm_AR488_ESP32S2_R4` layout and I see that it uses the pin mapping array method used by the custom layout. I haven't actually tested any of the ESP layouts myself yet, which is why there are no pre-defined ESP specific functions yet. The layouts have been pulled from David Douards AR488-ESP32 repository, but its sounds like you have and have been successfully using this configuration up to this point. In any case, testing the ESP32 was on my todo list. I don't think I have an S2, but I do have a couple of Devkit v1 WROOM's which should hopefully suffice for test purposes.

It seems unlikely that cdc changes would have affected reading data from the GPIB bus, but thank you for testing and ruling that out conclusively.

Thinking about it, this would probably have to be related to the ATN read loop but I need to dig deeper and investigate when  have some time.

WilheJo, 2025-07-05:
Still, something's fishy.
I get the correct response, but with debug enabled, I get the following log:

++ver
*IDN?
++read

AR488 GPIB controller, ver. 0.53.17 (JW), 03/07/2025
AR488_GPIBbus.cpp:593 (receiveData) > Start listen ->
AR488_GPIBbus.cpp:594 (receiveData) > Before loop flags:
AR488_GPIBbus.cpp:595 (receiveData) > TRNb: 0
AR488_GPIBbus.cpp:596 (receiveData) > rEOI: 0
AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
52 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
4F AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
48 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
44 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
45 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
26 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
53 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
43 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
48 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
57 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
41 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
52 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
5A AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
2C AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
53 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
4D AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
59 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
30 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
31 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
2C AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
38 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
33 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
39 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
2E AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
30 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
36 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
39 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
2F AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
30 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
30 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
32 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
2C AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
32 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
2E AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
30 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
32 AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!
0A AR488_GPIBbus.cpp:1229 (readByte) > Handshake error!

AR488_GPIBbus.cpp:687 (receiveData) > After loop flags:
AR488_GPIBbus.cpp:689 (receiveData) > TMO: 1200
AR488_GPIBbus.cpp:690 (receiveData) > Bytes read:  37
AR488_GPIBbus.cpp:691 (receiveData) > <- End listen.
AR488_GPIBbus.cpp:706 (receiveData) > Timeout waiting for sender!
AR488_GPIBbus.cpp:707 (receiveData) > Timeout waiting for transfer to complete!
AR488_GPIBbus.cpp:734 (receiveData) > done.

Twilight-Logic, 2025-07-13:
Both the ATN timing and the handshake error should be resolved in 0..53.20.