Twilight-Logic/AR488 #59: Add maximum_size parameter to receiveData()

hb020, 2025-05-04:
In my integration of the AR488 GPIB interface behind a VXI-11.2 frontend, I had to adapt `receiveData()`: 
* add a `maximum_size` parameter
* return the reason of the exit from the function (EOI, EndByte, Timeout, MaxSize,....).

Now this is not directly useful for Prologix, but it is a requirement for VXI-11. Are you willing to support that use case? If so, I am willing to provide a PR.

With the change, it works beautifully behind VXI-11.

Also, a separation of the Prologix part from the setup + loop part (now all mixed in AR488.ino) might be better, but that is not the point of this issue.

Twilight-Logic, 2025-05-05:
> In my integration of the AR488 GPIB interface behind a VXI-11.2 frontend, I had to adapt `receiveData()`:
> 
>     * add a `maximum_size` parameter

Apart from quitting reaceiveData(), What happens when maximum is reached? Will incoming data need to e buffered?

>     * return the reason of the exit from the function (EOI, EndByte, Timeout, MaxSize,....).

No problem with this. Just requires changing the return type from bool to uint8_t and assigning some values and then updating each statement where it is already used. Was already planning to do this at some point anyway so I might as well get it into the next release.

> Now this is not directly useful for Prologix, but it is a requirement for VXI-11. Are you willing to support that use case? If so, I am willing to provide a PR.
>
> With the change, it works beautifully behind VXI-11.

I would like to see how this has been implemented if possible please.

> Also, a separation of the Prologix part from the setup + loop part (now all mixed in AR488.ino) might be better, but that is not the point of this issue.

I think I can see how this might help with switching between Prologix and VXI-11. One would just switch the appropriate module. I will give that some thought.

hb020, 2025-05-06:
> Apart from quitting reaceiveData(), What happens when maximum is reached? Will incoming data need to e buffered?

The idea is to stop reading, leave the device addressed, and read the rest later. Stop and Go approach.

> I would like to see how this has been implemented if possible please.

See https://github.com/Kofen/PoE_Ethernet_GPIB_Adapter/blob/7-allow-large-reply-when-using-vxi/SW/src/AR488_GPIBbus.cpp, line 530

It seems that some device are not happy with the way I leave the reads, so I might need to change it a bit more, still investigating. Maybe more like what I did for the writes, where the same max size problem happens, but reversed: https://github.com/Kofen/PoE_Ethernet_GPIB_Adapter/blob/7-allow-large-reply-when-using-vxi/SW/src/main.cpp line 89 (the SCPI_handler class is the interface between VXI and the GPIB bus, in other projects I use that class to connect to a SCPI parser) 
But the HPAK material I tested is OK with how it works now. 

> I think I can see how this might help with switching between Prologix and VXI-11. One would just switch the appropriate module. I will give that some thought.

See https://github.com/Kofen/PoE_Ethernet_GPIB_Adapter/blob/7-allow-large-reply-when-using-vxi/SW/README.md

Twilight-Logic, 2025-05-08:
I have just pushed a new release and modified receiveData() to provide the return reason as you suggested. It has been done pretty much as in the code you linked, except that instead of adding a parameter, the return type of the function has been changed from bool to an enum, which allows multiple states to be returned rather than just OK or ERR. Otherwise the states are pretty much as you had set out, but using somewhat more descriptive names. I used the suffix `_LIMIT` but happy to change that to `_MAXSIZE` (or just `_MAX`?) if you prefer.

I am happy for you to submit a PR for the MaxSize parameter when you are ready. I will also look at moving the Prologix functions into a separate module and separating them from the main .ino.

Twilight-Logic, 2025-07-18:
The proposed change for maxSize and isLastPacket has been added manually to the current version of the code so the pull request has been closed. However, then you for your submission. The requested changes and fix are now included in the latest version which has been uploaded to the repository.

I am working on a new re-factored version of the project and since the code and the approach to going idle has changed, these changes cannot be applied as they stand to this version. Going to idle is handled in a different way. The interface will go to idle only when an untalk or unlisten is sent. If one is not sent, then it stays active in the current mode. This alternative solution will need to be tested in a real world scenario where it matters such as yours. I am still considering whether it will be worthwhile to back-port it to the existing code. However, as applied to the existing code, this request is now complete.

closed 2026-01-17 by hb020