Twilight-Logic/AR488 #65: SDC must not cause a reboot

dl8dtl, 2025-06-01:
When trying to receive data from my R4136 spectrum analyzer as a plotter (i.e. GPIB device), this basically works in `++lon 1` mode.
However, being explicitly addressed (on listener address 5, hard-wired into the analyzer), it causes a reboot.

It turns out the analyzer sends to following command sequence before starting to plot:
```
UNL
TAD 30
LAD 5
SDC
```
About 150 ms later, it takes back the `ATN` signal, and another 100 ms later, it starts emitting plot data.

The `SDC`, however, causes the AR488 firmware to reboot. After that, it no longer responds (obviously, because it's in controller mode again).
To me, this looks completely wrong. SDC is meant for the device to clear any internal state, but it must not turn the AR488 into a controller again.
Ill side-effect on the ESP32-based hardware: as the communication channel is a CDC (communication device class), the reboot drops the device from the USB, causing anything on the host to disconnect. (This is also the case for `++rst`, but it would be more tolerable there.)

Twilight-Logic, 2025-06-01:
What is talk address 30? It sounds like the interface needs to be in device mode on address 5. What it needs is for the state to be saved (++savecfg) so that when it restarts it in device mode still. Is ++savecfg not saving the configuration parameters?

Not sure what can be done regarding CDC state on restart and that is exactly what ++rst does, but maybe SDC does not need to reboot the device, which I think SDC received in device mode does currently. Will need to think about that one.

dl8dtl, 2025-06-02:
> What is talk address 30?

The spectrum analyzer that's going to emit plot data.

> It sounds like the interface needs to be in device mode on address 5.

It is. Not in a saved configuration, but in the live one.

It would be good to have two different savable addresses for controller and device mode, but that's a different story (in controller mode, I wouldn't want to use address 5).

> Not sure what can be done regarding CDC state on restart and that is exactly what ++rst does, but maybe SDC does not need to reboot the device, which I think SDC received in device mode does currently.

Exactly my point. ;-)

Twilight-Logic, 2025-07-03:
SDC fix is now included in the latest release. It no longer calls rst_h().