Twilight-Logic/AR488 #66: error: 'strlcpy' was not declared in this scope

hb9dug, 2025-06-20:
With an unused **Ardino UNO R4 Minima**, I wanted to try out this great AR488 project.
I installed Ardino IDE 2.3.6 and loaded AR488-0.5.11. Unfortunately the check ended with :
Compilation error: “strlcpy” was not declared in this scope._

After searching on the net, it seems that this function is not available on UNO R4...  Continuing my research, I found the code below which I integrated into AR488.ino.

#include <Arduino.h>
// Implementation of strlcpy for Arduino
size_t strlcpy(char *dst, const char *src, size_t dstsize) {
  size_t len = 0;
    // Check if destination size is greater than zero
  if (dstsize > 0) {
    while (src[len] != '\0' && len < dstsize - 1) {
      dst[len] = src[len]; // Copy character
      len++;
    }
    dst[len] = '\0'; // Null-terminate the destination string
  }
  // Calculate the length of the source string
  while (src[len] != '\0') {
    len++;
  }
  return len;
}

Unfortunately, compilation ended with the following error msg:
_...\Temp\ccEdzlwD.s: Assembler messages:
...\Temp\ccEdzlwD.s:36: Error: bad instruction `jmp 0'
...\Temp\ccEdzlwD.s:7388: Error: bad instruction `jmp 0'
exit status 1_

Any help in solving this issue will be welcome.

Best regards, Michel

Twilight-Logic, 2025-06-20:
Thank you for the report.

I installed the R4 library and got the same error regarding strlcpy. In this case, since we are copying a set number of bytes, the function can be replaced with strncpy with a little adjustment. Or, as you have done, you can use a loop to copy the characters. I will update this in the next release.

The `bad instruction jmp 0` error relates to the rst_h function. The function has conditional compilation implemented for the various types of boards, or more specifically for the MCU that it has board. If it doesn't find a match it defaults to the AVR version. However, the R4 has a Renesas RA4M1 ARM processor. I need to find and test the correct reset call for this.

I also found when I did a test compile, that there were errors relating to EEPROM that need to be fixed. I need to read up, but I suspect that although the R4 is classed as an AVR board, it does not have EEPROM in the traditional sense and its likely that saving settings needs to be handled in a file using the LittleFS filesystem like it does on an ESP32. I will update this issue when I have researched these problems and have a working solution.

Twilight-Logic, 2025-06-20:
I think if you add the following after line 1633, it should fix the `bad instruction jmp 0` error:

```
#elif defined(ARDUINO_UNOR4_MINIMA) || defined (ARDUINO_UNOR4_WIFI)
  NVIC_SystemReset();
```
Upon further investigation it seems that the R4 does have EEPROM and the standard Arduino EEPROM library should work. However, I had to add a guard in two places to get a compile. This is the modified default_h function:

```
void default_h(char *params) {
  #if defined(__AVR__) || defined (ESP32)
  if (params != NULL) {
    if (strncasecmp(params, "wipe", 4) == 0) {
      epErase();
      dataPort.println(F("EEPROM erased!"));
    }else{
      errorMsg(2);
    }
  }
  #endif
  gpibBus.setDefaultCfg();
}
```
Then also in AR488_EEPROM.h:

```
#ifndef AR488_EEPROM_H
#define AR488_EEPROM_H

#include "AR488_Config.h"
//#include <EEPROM.h>

/***** AR488_Eeprom.h, ver. 0.01.03, 28/04/2025 *****/

/*
 * EEPROM SIZES:
 * 
 * ATmega2560/1284   4096   // Mega 2560, MightyCore 1284 
 * ATmega644         2048   // MightyCore 644
 * ATmega328/32u4    1024   // Uno, Nano, Leonardo
 * ATmega168          512
 * ATmega4809         256   // Nano Every
 * ESP8266/ESP32     none   // Emulated. 512kb assigned?
 */

#if defined (__AVR__) || defined (ESP32)

.
.

#endif  // defined (__AVR__) || defined (ESP32)

#endif  // AR488_EEPROM_H
```
I don't have the board to test, but this does at least allow the code to compile.
I would be interested to know whether it works as expected!

Twilight-Logic, 2025-06-20:
BTW, I modified setvstr_h as follows:

```
void setvstr_h(char *params) {
  uint8_t plen = strlen(params);
  char idparams[64];
  const char *vstr = "verstr ";

  if (plen > 47) plen = 47;
  memset(idparams, '\0', 64);
  strncpy(idparams, vstr, 7);
  strncat(idparams, params, plen);

  id_h(idparams);
}
```
I have tested it on my UNO R3, but would appreciate it if you could confirm it works on the UNO R4.

hb9dug, 2025-06-21:
Thank you for your reply.

In sequence, I have :
- modified the setvstr_h function, no more ‘strlcpy’ error but compilation leaves me with the ‘jump 0’ error

- modified AR488.ino and AR488_EEPROM.h, no additional errors, but compilation gives me the ‘jump 0’ error

- in my AR488.ino, line 1633 doesn't make sense, so I inserted the 2 lines here:

void rst_h() {
#ifdef WDTO_1S
  // Where defined, reset controller using watchdog timeout
  unsigned long tout;
  tout = millis() + 2000;
  wdt_enable(WDTO_1S);
  while (millis() < tout) {};
  // Should never reach here....
  if (isVerb) {
    dataPort.println(F("Reset FAILED."));
  };
#elif defined (ESP32)
  ESP.restart();
#elif defined(ARDUINO_ARCH_RP2040)
  rp2040.reboot();
**#elif defined(ARDUINO_UNOR4_MINIMA) || defined (ARDUINO_UNOR4_WIFI)
  NVIC_SystemReset();**
#else
  // Otherwise restart program (soft reset)
  asm volatile ("  jmp 0");
#endif
}

but where **ARDINO_UNOR4_MINIMA** is defined ?

I still tried to compile but without success with 40-50 lines of error...

That's the status of things
Cordialement,
Michel

hb9dug, 2025-06-21:
SUPER as we say in French !  Compiled without a glitch !

All that remains now is to wire an HP-IB connector recovered from an HP instrument chassis.

I'll keep you posted on my work as I go along.

Merci beaucoup !

Cordialement,
Michel