Twilight-Logic/AR488 #68: ++savecfg not working for ESP32

g-user, 2025-06-26:
++savecfg doesn´t work correctly for 2 reasons:
1.  "E2END" is used as a flag for existing/nonexisting EEPROM, but this is not defined in the ESP32 (emulated) EERPOM library
2.  in the epWriteData/epReadData the variable is a pointer to the buffer (uint8_t cfgdata])  but then "EEPROM.put(addr,cfgdata);" is used for saving the data. So not the buffer is saved but the pointer to the buffer. This is not right. So better use 

for (i=0; i<cfgsize; i++){
    EEPROM.write(i+EESTART,cfgdata[i]);

see PR " correct EEPROM read/write for ESP32 #67 "

Twilight-Logic, 2025-06-27:
Thank you for raising this issue. xFunnily enough I was looking at EEPROM last night in conjunction with a Mega 4809 and came across a statement that said that E2END might not be defined for some boards. This had worked in the past. Maybe this changed in Espressif Arduino core3? A lot of things did. In AR488-ESP32, David Douard used the file LittleFS filesystem  instead and wrote the settings to a file which might be a better approach on the ESP. 

I am unsure in your second point why EEPROM.put() is saving a pointer instead of the object. The array of uint8_t values is passed by reference (uint8_t cfgdata[]) so EEPROM.put() should be treating this as an array object, which it does for AVR boards. The .put() function is supposed to take care of the iteration. Whatever the case, saving the pointer is indeed not right and needs to be fixed and I guess that manually iterating the array and using .write() to write the individual bytes is the obvious workaround.

g-user, 2025-06-27:
thanks for the fast merge!  I´m pretty sure there are exist more elegant methods for solving this, but it was  the easiest fix (for me).

closed 2025-06-27 by g-user